lgsvl/simulator #462: Simulator does not start on Ubuntu Mate 18.04

RamoramaInteractive, 2019-11-03:
Beside Windows 10 I also tried to run the LGSVL Simulator on Ubuntu Mate 18.04 computer.
According system info my computer has the following data:
OS: Ubuntu Mate 18.04
Intel(R) Core(TM) i7-4790 CPU @ 3.60GHz
Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller (8 GB).

If I execute the simulator from https://github.com/lgsvl/simulator/releases/tag/2019.10-rc1
nothing happens.

I already installed Unity3D on Ubuntu, but I only receive this lines on my terminal:

```
Set current directory to /home/hsfulda/Desktop/lgsvlsimulator-linux64-2019.10-rc1
Found path: /home/hsfulda/Desktop/lgsvlsimulator-linux64-2019.10-rc1/simulator
Mono path[0] = '/home/hsfulda/Desktop/lgsvlsimulator-linux64-2019.10-rc1/simulator_Data/Managed'
Mono config path = '/home/hsfulda/Desktop/lgsvlsimulator-linux64-2019.10-rc1/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Preloaded 'ScreenSelector.so'
Display 0 'DELL P2314H 23"': 1920x1080 (primary device).
Loading player data from /home/hsfulda/Desktop/lgsvlsimulator-linux64-2019.10-rc1/simulator_Data/data.unity3d
Logging to /home/hsfulda/.config/unity3d/Editor/Player.log
```

martins-mozeiko, 2019-11-03:
Please check the log file at `~/.config/unity3d/Editor/Player.log` location. It should have more detailed error message at bottom of the file.

One reason could be that you don't have installed Vulkan user-space libraries and/or driver. Check if correct packages are installed: https://www.lgsvlsimulator.com/docs/faq/#why-does-the-simulator-not-open-on-linux

RamoramaInteractive, 2019-11-04:
> Please check the log file at `~/.config/unity3d/Editor/Player.log` location. It should have more detailed error message at bottom of the file.
>
> One reason could be that you don't have installed [...]

I had installed Vulkan, but according to the logs Vulkan seems to be not properly installed:
```
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=2
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan error VK_ERROR_INCOMPATIBLE_DRIVER (-9) file: ./Runtime/GfxDevice/vulkan/VKContext.cpp, line: 333
Vulkan error./Runtime/GfxDevice/vulkan/VKContext.cpp:333
Vulkan detection: 0
No supported renderers found, exiting
 
(Filename: ./PlatformDependent/LinuxStandalone/main.cpp Line: 639)
```
The link to the instruction of Vulkan you posted here is the exact same link I'd opened when I tried to install Unity on Ubuntu. Maybe my Ubuntu computer does not support any Graphic driver. :(

martins-mozeiko, 2019-11-04:
This error probably means that your GPU does not support Vulkan API.

RamoramaInteractive, 2019-11-04:
> This error probably means that your GPU does not support Vulkan API.

That's a pity. Is there anything I can do to run Vulkan API or alternative APIs? I was thinking to change the video card.

RamoramaInteractive, 2019-11-04:
I've tested the other instruction for installing (and enabling) Vulkan on Ubuntu:

- https://linuxconfig.org/install-and-test-vulkan-on-linux
- https://software.intel.com/en-us/articles/enabling-vulkan-vk-intel-performance-query-extension-in-ubuntu

The Simulator starts very slowly, but it should work for now.

martins-mozeiko, 2019-11-04:
Intel GPUs are not really meant applications with heavy GPU usage. While they might support basic functionality, simulation requires a lot of GPU resources to render high quality environment, and perform high performance camera, lidar & other sensor rendering. Out simulator main feature is to have very realistic rendering and high quality sensor simulation.

RamoramaInteractive, 2019-11-04:
> I strongly suggest using Nvidia or AMD GPU.

Thank you for your advice. I look to a computer which supports these kind of GPU.
Yes, actually Intel GPUs are not the best choice.

RamoramaInteractive, 2019-11-05:
I open the `cat ~/.config/unity3d/Editor/Player.log` again:
```

Unloading 5 Unused Serialized files (Serialized files now loaded: 3)
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp, line: 1197
Vulkan error./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp:1197
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKSwapChain.cpp, line: 626
Vulkan error./Runtime/GfxDevice/vulkan/VKSwapChain.cpp:626
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp, line: 1197
Vulkan error./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp:1197
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKSwapChain.cpp, line: 626
Vulkan error./Runtime/GfxDevice/vulkan/VKSwapChain.cpp:626
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp, line: 1197
Vulkan error./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp:1197
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp, line: 1197
Vulkan error./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp:1197
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp, line: 1197
Vulkan error./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp:1197
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp, line: 1197
Vulkan error./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp:1197
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp, line: 1197
Vulkan error./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp:1197
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp, line: 1197
Vulkan error./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp:1197
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp, line: 1197
Vulkan error./Runtime/GfxDevice/vulkan/VKTaskExecutor.cpp:1197
Vulkan error VK_ERROR_DEVICE_LOST (-4) file: ./Runtime/GfxDevice/vulkan/VKSwapChain.cpp, line: 626
Vulkan error./Runtime/GfxDevice/vulkan/VKSwapChain.cpp:626
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'IndirectDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'ForwardDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'VisibilityDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
Receiving unhandled NULL exception

```
I hope I can change the computer which supports Nvidia or AMD GPU.

closed 2020-04-08 by EricBoiseLGSVL