lgsvl/simulator #464: Unity Editor is freezing on play mode when pedestrian is added.

clepz, 2019-11-04:
Hi everyone,
I tried to add pedestrian path for my map using map annotation tool. However, Unity Editor is freezing on play mode after adding the pedestrian path. Screenshots about this problem is added.
How can I overcome this problem. Thank you.

[image]

[image]

[image]

shalinmehtalgsvl, 2019-11-04:
Hi @clepz ,

I was able to recreate your issue and the problem is that we did not include a NavMesh with Shalun (will be updated in next push). To generate your own here are the steps:

1. Open AI > Navigation menu
[image]

2. Select the Meshes that you want to include in the NavMesh. (In this screenshot I have only included the sidewalks). Check the `Navigation Static` box and then select the type of area the mesh is. You can select multiple meshes at the same time, but they will all be marked as the same type of area (so do all sidewalks, then do all roads, then do all grass, ...). 
[image]

3. Bake NavMesh. After baking, the NavMesh will be displayed in the scene view. In the screenshot I only baked a NavMesh on the sidewalks (green polygons). If you have set other meshes, they will be colored based on the `Navigation Area` set for the mesh. 
[image]

clepz, 2019-11-05:
Thank you. It works perfectly.

closed 2019-12-29 by clepz