lgsvl/simulator #465: Can't Compile Project in unity 2019.1.10f1

yansvai, 2019-11-04:
problems in compiling project in unity

Assets\Scripts\Bridge\Cyber\CyberWriter.cs(12,7): error CS0246: The type or namespace name 'ProtoBuf' could not be found (are you missing a using directive or an assembly reference?)

C:\Users\user\Documents\simulator-2019.10-rc1\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipeline\HDRenderPipeline.cs:395 

  at UnityEngine.Experimental.Rendering.HDPipeline.HDRenderPipeline..ctor (UnityEngine.Experimental.Rendering.HDPipeline.HDRenderPipelineAsset asset) [0x0033c] in C:\Users\user\Documents\simulator-2019.10-rc1\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipeline\HDRenderPipeline.cs:262 

  at UnityEngine.Experimental.Rendering.HDPipeline.HDRenderPipelineAsset.CreatePipeline () [0x00004] in C:\Users\user\Documents\simulator-2019.10-rc1\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipeline\HDRenderPipelineAsset.cs:35 
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

RamoramaInteractive, 2019-11-04:
I've got the same problems, but under Ubuntu. Strangely under Windows the compiling problems disappeared. Can you at least start the LGSVL Simulator?

david-gwa, 2019-11-11:
mabe the download files are unfinished, try to git pull the whole project again and rebuild it

martins-mozeiko, 2019-11-11:
Most likely you have not installed [Git LFS](https://git-lfs.github.com/). We use Git LFS to store large binary files to speed up clone process. Make sure you install before doing git clone. Or do `git lfs pull && git lfs checkout` afterwards to fix repository.

https://www.lgsvlsimulator.com/docs/faq/#why-are-assets-scenes-missing-empty-after-cloning-from-git

closed 2020-04-08 by EricBoiseLGSVL