lgsvl/simulator #466: MapOrigin Editor Rounding Issue

moenasrallah, 2019-11-04:
When Origin Easting and Northing are entered in the editor for MapOrigin, it is converting from string (values entered in the editor) to double (values in the maporigin script).

This is rounding the UTM coordinates to inaccurate values, causing bad and unnecessary offsets. I believe the solution for this is to take in the input from the editor as a string, and then do proper conversion to double so no rounding happens.

This is especially true with Northing (larger value)

moenasrallah, 2019-11-04:
Example: Entering 4321398.52 for either Easting or Northing will appear as 4321399

martins-mozeiko, 2019-11-04:
I think you mean that it is rounded to float, not double. as OriginEasting and OriginNorthing is floats inside MapOrigin component.

This is already reported in #463. We'll change it to double.

moenasrallah, 2019-11-04:
Yes, sorry, I edited it. Sorry for posting a duplicate issue. I will move comments onto there and close this

closed 2019-11-04 by moenasrallah