lgsvl/simulator #467: Reading topics between of the LG simulator

mkhayatian, 2019-11-04:
Reading ROS topics apollo/canbus/chassis outside the docker (in another platform connected to the ros master) requires defining the "pb_msgs" type. Is there a way to add it as a standalone package?

martins-mozeiko, 2019-11-04:
Using pb_msgs may require to modify ROS. Apollo did modifications to ROS protocol & libraries to support protobuf encoding for messages. I'm not sure if there is easy way to get it in standard ROS environment. I suggest asking this in Appllo issues: https://github.com/ApolloAuto/apollo/issues

mkhayatian, 2019-11-06:
I did a bit of search and found that pb_msg type is defined by the LG simulator bridge. Is there a way to subscribe and publish a message to those topics?

martins-mozeiko, 2019-11-06:
pb_msgs is defined by Apollo. They modified ROS protocol to allow sending protobuf encoded messages, not only standard ROS types. You'll need to use ROS environment provided by Apollo if you want to use pb_msgs types.

mkhayatian, 2019-11-06:
Ok. Is there any way to read and write to messages between LG simulator and Apollo?

martins-mozeiko, 2019-11-06:
It depends how you want to send these messages.

If you want to read/write them in external scripts (your code) outside of Docker container, then you cannot use pb_msbs directly. You need to either run your code inside Apollo ROS environment, or communicate with rosbridge directly - which is how Simulator is talking to Apollo. Simulator does not send pb_msgs as ROS message directly. It communicates to Apollo through [rosbridge](http://wiki.ros.org/rosbridge_suite). We do this because we don't want to or cannot have full ROS environment running where Simulator is running - for example, Windows does not support ROS1 officially. If you want to go rosbridge route, then you need to read up and use [rosbridge protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md).

But if you want to send extra messages from Simulator code itself, then you need to read up how we implement message sending. Good overview for this is in #451 issue.

mkhayatian, 2019-11-06:
Thank You. very helpful.

mkhayatian, 2019-11-07:
I was able to establish a websocket and subscribe to /apollo/control topic. However, I cannot get sensor data that are published by the simulator e.g. /apollo/sensor/gnss/imu. Any suggestions?

martins-mozeiko, 2019-11-07:
What do you mean that you cannot get data published? Can you explain what exactly are you running?

mkhayatian, 2019-11-07:
I'm running a rosbridge client (outside docker) to connect to the rosbride websocket (inside docker "./script/rosbridge.sh").
When I define a subscriber in the client for reading data of a sensor topic e.g. "apollo/sensor/gnss/imu", I get nothing. However, I can read data of the "/apollo/control" topic .

martins-mozeiko, 2019-11-07:
Ah, I see. Rosbridge unfortunately does not send messages between its clients. It only forwards messages between its clients and ROS. So you can send message to ROS, you can receive from ROS. But you cannot send message from one client to another.

closed 2020-04-08 by EricBoiseLGSVL