lgsvl/simulator #468: You can export .bin file, is there a way of importing a .bin file into unity?

ninjacoder0, 2019-11-05:
Hi all,

I was wondering is there a method you can open a .bin file of a map in unity?
Basically reverse engineering it to be able to open it & modifying the map?

Thanks all

shalinmehtalgsvl, 2019-11-05:
Yes, you can import a .bin file into the simulator. This doc needs to be updated to reflect it, but it's the same process as importing a LaneLet2 file. 

https://www.lgsvlsimulator.com/docs/map-annotation/#import-map-annotations

martins-mozeiko, 2019-11-05:
However take into account that import/export process is not lossless. During the import we import only things that are necessary for our simulation. We will ignore all other information in hd map file. Also existing information like lanes could be changed significantly if necessary. They will still represent same information, but data could look very different.

Simulator is not designed to be an editor for hd vector map.

daohu527, 2019-11-12:
I agree with martins-mozeiko.   
The simulator is concerned with not only maps but also models. For example, Maybe "ClearArea" is not rendered in the simulator as a model.
At present, the simulator is still concerned with obtaining the necessary map information from the hd map. If you need to edit the hd map, it is recommended to use other tools. 
In general, these two have different requirements for the map.

closed 2020-04-08 by EricBoiseLGSVL