lgsvl/simulator #469: How to change the car model into motorbike model for autodriving

clouddd9, 2019-11-06:
Hello, I'm a beginner and using lgsvl&apollo5.0 for autonomous driving simulation. However, there are some changes in my LAB's experiments. Our entity prototype is a motorbike, so I need to change the default model into motorbike model. And I find it extremely difficult to modify the simulator code. therefore I hope you could help me. Thank you very much!

Q1: Does lgsvl using canbus to control the car? (OR What type of message does apollo5.0 send to lgsvl via cyber_bridge to control the car in the simulator?) If so, does that mean I need to add a new canbus model of the motorbike in both apollo5.0 and lgsvl?

Q2:I've looked for the Jaguar2015XE in Unity3D, and I notice that I need to change the following scripts:
<code>/Components/VehicleActions</code>
<code>/Components/VehicleDynamics</code>
<code>/Controllers/VehicleController</code>
what else codes in lgsvl simulator should I change for using motorbike if I've got the motorbike models(lights and indicators are not necessary in our experiments)?

clouddd9, 2019-11-07:
can anyone help me......

OutOfCage, 2019-11-07:
I have done the work about car model. Let me try to answer.
- For Q1, Apollo send control command to Simulator(lgsvl). Simulator send the data of Localization, Perception and Canbus to Apollo. You can see the files of `/Bridges/` and `/Sensors`.
`/Bridges/Data/VehicleControlData.cs` and `/Sensors/VehicleControlSensor.cs` recieve control data from Apollo.  VehicleControlData is converted from apollo.control.ControlCommand(`/Bridges/Cyber/CyberConversions.cs`).

- For Q2, these issues #401 #441 are about car model. I think adjusting parameters of VehicleDynamics in Unity is enough. What Apollo Control care is relationship between acceleration and control cammands. For example, my vehicle's speed is 10m/s and I hope it has 1m/(s*s) acceleration.  What throttle I need to send?  You can see the calibration table in this [file](https://github.com/lgsvl/apollo-5.0/blob/simulator/modules/calibration/data/Jaguar2015XE/control_conf.pb.txt).

clouddd9, 2019-11-07:
Thank you for your help!!! I've tried to edit my motorcycle in unity3D based on Jaguar2015XE following the steps #401 . 
[image]
Do you mean that all I need to do are
1.adjusting the parameters of <code>VehicleDynamics</code>;
2.copy and adjust the Lightsgroup of Jaguar2015XE to my motorcycle model to avoid adjust the <code>VeicleActions</code> (which is the easiest way for me......);
to match my motorcycle model? 
Besides, I've been confused about How does the canbus work with cyber_bridge. Here's my understanding:

the lgsvl simulator doesn't send the message of the type of the car model to Apollo5.0. It just sends normal canbus info and perception&localization messages to Apollo. It's in Apollo that we choose the type of car model and then Apollo makes planning based on the input messages(especially the corresponding vheicle prototype in <code>/apollo5.0/modules/canbus/vehicle</code> ). So if I want to make the autodriving simulation of my own motorcycle, I still need to create a motorcycle canbus prototype just like Lincoln, Lexus, gem and so on. (and, yes, another set of planning, control algorithms...But that's no the point now)

Am I wrong? Hope you can give me some advice, thank you!

OutOfCage, 2019-11-07:
- I think first you can make a model that can run in map under manual control. WheelCollider is important in Dynamics and you need to change correctedly. Also your Motorcycle model needs mesh collider. Meeting these conditions, vehicle should be able to spawn. 
- For canbus, the data type of chassis can meet your demand? If it is enough, you can just change in Apollo. If you need to add a new data type to send messages between Apollo and Simulator, you can add a new sensor to do this work.
- In Simulator, vehicle's movement is controlled by Dynamics after recieving control commands. You can see it and then consider a solution.

closed 2019-11-29 by clouddd9