lgsvl/simulator #472: understanding gear shifting and acceleration in VehicleControl.cs file

Aquarian-source, 2019-11-06:
use case: autoware message types to connect to lg sim using ROS1
i have few questions: 
1. for accelerating and braking i am using ctrl_cmd.linear_acceleration. is this the right practice?
2. what is significance of gears in this. for acc gear = 64 and for braking gear = 1.  what is the understanding for gears?
3. also what does this mean Dynamics.GearboxShiftUp()" or "Dynamics.GearboxShiftDown()" ?
4. how is steering done in this case?
5. do we have linear acceleration in lg sim car model? 

Thank you in advance. you guys are a big help in understanding the lg simulator. Cheers

martins-mozeiko, 2019-11-07:
1) Yes, braking and acceleration is handled by seting linear_acceleration value from -1 to +1

2) If gear has value 64 then simulator shifts gear up, otherwise if gear has nonzero value, it will shift gear down. 0 will mean no changes to gears

3) This is part or vehicle dynamics, that makes car to use higher or lower gear.

4) Steering is handled by settings ctrl_cmd_steering_angle value as degrees.

5) You cannot set linear acceleration directly to rigid body of vehicle. You can only apply throttle/braking values.

Aquarian-source, 2019-11-07:
i understood everything. for last question may be i asked the question wrongly. i wanted to ask through throttling or braking can a car accelerate non-linearly in Lg simulator?

also from VechicleDynamics.cs file i understood LG simulator changes the gear depending on the RPM and a predefined curve. is this right? can you please elaborate this how we are changing gears

martins-mozeiko, 2019-11-07:
What do you mean by non-linear acceleration?

When you send throttle value to Simulator, that is translated to torque that is applied to wheels. And wheels move car forward, that's how you perceive acceleration and velocity of vehicle.

closed 2020-04-08 by EricBoiseLGSVL