lgsvl/simulator #473: docker issue

stehp, 2019-11-06:
When I run `./docker/scripts/dev_start.sh`, I get the following error. I am new to docker. How can I fix the error below? 

```
docker: Error response from daemon: OCI runtime create failed: container_linux.go:346: starting container process caused "process_linux.go:449: container init caused \"process_linux.go:432: running prestart hook 0 caused \\\"error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: initialization error: driver error: failed to process request\\\\n\\\"\"": unknown.

Error response from daemon: Container f748e6b1827497c67856ab5fe80ef566cc12f2c859029954fe161fcde5eb4270 is not running
```

I am using lgsvl's image: lgsvl/apollo-5.0:latest.

martins-mozeiko, 2019-11-06:
Please check first if nvidia runtime is available for your docker installation. Run `docker info | grep Runtimes` then you should see:

    Runtimes: nvidia runc

If you don't see nvidia there, that means nvidia-docker2 is not install correctly.

Next try running nvidia-smi inside container with nvidia-docker command:

    nvidia-docker run -ti --rm nvidia/cuda:9.0-base nvidia-smi

It should show GPU name and driver version in the table. If this does not work, then something is wrong with docker service or nvidia runtime. Try restarting docker service, or even reboot whole computer - sometimes this helps.

strahl2e, 2019-12-29:
@martins-mozeiko  For docker 19.03 and later nvidia-docker2 is deprecated so Docker runtimes don't show nvidia.  However, the restart solution did work :)

More info: https://github.com/NVIDIA/nvidia-docker

closed 2020-04-08 by EricBoiseLGSVL