lgsvl/simulator #474: Null-safe light disable

KrzysztofDabek, 2019-11-07:
Hi,

https://github.com/lgsvl/simulator/blob/336ce3a6b881d9ea23e7e826c5d5c551283c9c53/Assets/Scripts/Components/VehicleActions.cs#L270
In this line there is no null-safe operation as in all previous cases. Of course it is because **interiorLight** is not a list but a singular light and **?.** operator cannot be assigned in C#.
This inconsistency should not cause any trouble if we are building a regular car with interior light.
But as a part of my project I wanted to build a container for sensors without that object and because it was null an error was thrown. 

`if(interiorLight != null)
            interiorLight.enabled = false;`
My suggestion is to add a check for null here as is in case of other light objects. A good idea could be also to change type of interiorLight to List<Light>. Therefore, it is both possible to add more than one light inside of a car and the code is consistent doing a null safe operation

EricBoiseLGSVL, 2019-11-07:
@KrzysztofDabek 
This would be an easy change.  We will add on the next release.

EricBoiseLGSVL, 2019-11-19:
@KrzysztofDabek fix will be in the next release.  Thanks again for pointing this issue out

closed 2020-04-08 by EricBoiseLGSVL