lgsvl/simulator #475: AssetBundle build issue with new Simulator version 2019.10

Roman-jj, 2019-11-07:
Hi, after updating to the newest version on Windows, I can't build AssetBundles anymore.

I saw that now it should be possible to build for cross-platform support, but when I try to build I first get a warning saying
> Library\PackageCache\com.unity.inputsystem@0.2.10-preview\InputSystem\Plugins\PlayerInput\PlayerInput.cs(399,27): warning CS0109: The member 'PlayerInput.camera' does not hide an accessible member. The new keyword is not required.

then an error

> Building an AssetBundle for target 'LinuxStandaloneSupport' is not allowed because the required module is not installed.
UnityEditor.BuildPipeline:BuildAssetBundles(String, AssetBundleBuild[], BuildAssetBundleOptions, BuildTarget)
Simulator.Editor.Build:RunAssetBundleBuild(String, List`1, List`1) (at Assets/Scripts/Editor/Build.cs:397)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:165)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

and another error

> FileNotFoundException: Could not find file "C:\Users\joerger\LGSVL\simulator\AssetBundles\e5fd4294-d90b-45f1-a0c0-27e54437b526_vehicle_main_linux"
System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) (at <7d97106330684add86d080ecf65bfe69>:0)
System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share) (at <7d97106330684add86d080ecf65bfe69>:0)
(wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare)
System.IO.File.Open (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share) (at <7d97106330684add86d080ecf65bfe69>:0)
ICSharpCode.SharpZipLib.Zip.StaticDiskDataSource.GetSource () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile+ZipUpdate.GetSource () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile.AddEntry (ICSharpCode.SharpZipLib.Zip.ZipFile workFile, ICSharpCode.SharpZipLib.Zip.ZipFile+ZipUpdate update) (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile.RunUpdates () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile.CommitUpdate () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
Simulator.Editor.Build.RunAssetBundleBuild (System.String folder, System.Collections.Generic.List`1[T] environments, System.Collections.Generic.List`1[T] vehicles) (at Assets/Scripts/Editor/Build.cs:413)
Simulator.Editor.Build.OnGUI () (at Assets/Scripts/Editor/Build.cs:165)
System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) (at <7d97106330684add86d080ecf65bfe69>:0)
Rethrow as TargetInvocationException: Exception has been thrown by the target of an invocation.
System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) (at <7d97106330684add86d080ecf65bfe69>:0)
System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) (at <7d97106330684add86d080ecf65bfe69>:0)
UnityEditor.HostView.Invoke (System.String methodName, System.Object obj) (at C:/buildslave/unity/build/Editor/Mono/HostView.cs:340)
UnityEditor.HostView.Invoke (System.String methodName) (at C:/buildslave/unity/build/Editor/Mono/HostView.cs:334)
UnityEditor.HostView.InvokeOnGUI (UnityEngine.Rect onGUIPosition, UnityEngine.Rect viewRect) (at C:/buildslave/unity/build/Editor/Mono/HostView.cs:310)
UnityEditor.DockArea.DrawView (UnityEngine.Rect viewRect, UnityEngine.Rect dockAreaRect, System.Boolean floatingWindow, System.Boolean isBottomTab) (at C:/buildslave/unity/build/Editor/Mono/GUI/DockArea.cs:373)
UnityEditor.DockArea.OldOnGUI () (at C:/buildslave/unity/build/Editor/Mono/GUI/DockArea.cs:340)
UnityEngine.UIElements.IMGUIContainer.DoOnGUI (UnityEngine.Event evt, UnityEngine.Matrix4x4 parentTransform, UnityEngine.Rect clippingRect, System.Boolean isComputingLayout, UnityEngine.Rect layoutSize) (at C:/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:295)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e, UnityEngine.Matrix4x4 worldTransform, UnityEngine.Rect clippingRect) (at C:/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:481)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e) (at C:/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:464)
UnityEngine.UIElements.IMGUIContainer.HandleEvent (UnityEngine.UIElements.EventBase evt) (at C:/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:444)
UnityEngine.UIElements.MouseCaptureDispatchingStrategy.DispatchEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at C:/buildslave/unity/build/Modules/UIElements/Events/MouseCaptureDispatchingStrategy.cs:93)
UnityEngine.UIElements.EventDispatcher.ProcessEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at C:/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:280)
UnityEngine.UIElements.EventDispatcher.Dispatch (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel, UnityEngine.UIElements.DispatchMode dispatchMode) (at C:/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:156)
UnityEngine.UIElements.BaseVisualElementPanel.SendEvent (UnityEngine.UIElements.EventBase e, UnityEngine.UIElements.DispatchMode dispatchMode) (at C:/buildslave/unity/build/Modules/UIElements/Panel.cs:189)
UnityEngine.UIElements.UIElementsUtility.DoDispatch (UnityEngine.UIElements.BaseVisualElementPanel panel) (at C:/buildslave/unity/build/Modules/UIElements/UIElementsUtility.cs:255)
UnityEngine.UIElements.UIElementsUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at C:/buildslave/unity/build/Modules/UIElements/UIElementsUtility.cs:78)
UnityEngine.GUIUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at C:/buildslave/unity/build/Modules/IMGUI/GUIUtility.cs:179)

I can get around the error by commenting out lines 391-401 and line 410 in build.cs, so it's not an issue for me, but I'm not sure if it's a bug or just some requirements that I don't have installed on my system

martins-mozeiko, 2019-11-07:
We recently changed how we build map/vehicle bundles. Before they were platform dependent - Windows vs Linux. Now they are built both at same time into one package. That means your Unity installation needs to have Windows and Linux support installed - you choose this during installation. You should simply run installation again and choose to install Linux support. Then build will succeed.

Roman-jj, 2019-11-07:
Ah thanks, then I'll just do that.

closed 2019-11-07 by Roman-jj

riveranb, 2019-11-09:
Could you update doc of build from source instructions ? 
https://github.com/lgsvl/simulator/blob/master/Docs/docs/build-instructions.md
Mention that installing Unity with necessary Windows & Linux build supports.

gaohongfein, 2020-03-20:
how to install linux support in windows unity?

Erdemhan, 2020-03-27:
> how to install linux support in windows unity?

Unity Hub > Installs > "Choose your version and click add modules " > Linux build support