lgsvl/simulator #476: Localization in Apollo and LGSVL

ningfeiwang, 2019-11-09:
Hello,

I found out that Apollo 5.0 posted a NDT localization module. When I try to use this localization and bridge it with LGSVL, I found out that there is no local map for this module. So, I wonder whether LGSVL provides some local map in your map(e.g., CubeTown, SingleRoad) so that I can take another localization module and simulate in LGSVL.

BTW, I find that LGSVL only uses RTK as localization. So, I wonder whether LGSVL will provide other options.

Thank you very much.

daohu527, 2019-11-12:
You mean provide **point cloud map** for SingleRoad or CubeTown ?  I think this maybe a good idea.

SingleRoad has nothing but roads, so the "point cloud map" is not available, I suggest you can try other maps.

ningfeiwang, 2019-11-12:
Thanks. I noticed that LGSVL released new maps like shalun. So, I wonder whether they will release some local maps for MSF or NDT localization in Apollo 5.0.

martins-mozeiko, 2019-11-12:
We've never tried NDT localization for Apollo. When we started with Apollo it was on version 3.0 which did not have ndt localization. If there is some extra things that are required from simulator side, we would be happy to implement those.

We know for sure that NDT localization works with Autoware on our current maps. For Autoware we generate point cloud with `Simulator -> Export Point Cloud` menu item. It will simulate lidar driving around the map and will collect points that will be saved to .pcd file. This very fast and automatic process which does not involve ROS/Cyber and parsing messages.

But if you need point scans in cyber bag for Apollo NDT localization, then with current implementation you'll need to set up simulator with Apollo and CyberBridge and drive around the map manually.

ningfeiwang, 2019-11-12:
Thank you. So, can I just use the .pcd file in Autoware and generate the local map for Apollo NDT? I thought the format of .pcd file is similar.

martins-mozeiko, 2019-11-13:
If Apollo can use it, then yes. We have not tried to do NDT localization, so I don't know what Apollo expects.

lemketron, 2020-09-25:
We know some users have managed to generate local map and enabled MSF for Apollo 5.0. 

However, the same approach does not work for Apollo master (and pre6 branch). We understand that some users are now trying to make it work with Apollo master (aka Apollo 6.0). (Thanks, @rongguodong!)

closed 2020-10-20 by EricBoiseLGSVL