lgsvl/simulator #477: fail to launch simulator

simon13579, 2019-11-10:
Hello,

I am using Ubuntu 16.04 in VMware, and I build the simulator from source code following the instructions, in the Unity output window some error shows:
[image]

However Simulator.exe is still generated. I double click the .exe, the Simulator Configuration dialog shows and I click OK, then nothing happen, the simulator didn't show in browser.
[image]

I also try to use the released executable from GitHub repository, but the problem is same. Is there anything more I can try?

Thanks.

martins-mozeiko, 2019-11-11:
The error from Editor looks like it run out of memory when building code. Do you have allocated enough memory for your VM?

As for running binary, you should check logfile at `~/.config/unity3d/Unity/Editor/Editor.log` location. It should have more detailed information.

My guess would be that it cannot properly use GPU because of VM or drivers. Check that you have configured your VM correctly. Run any vulkan application to see if can show something. For example, `vulkaninfo` (on Ubuntu it is inside `vulkan-utils` package). Verify that it outputs something like this with information about your GPU:

```
GPU0:
VkPhysicalDeviceProperties:
---------------------------
	apiVersion     = 4198509 (1.1.109)
	driverVersion  = 1824866304 (0x6cc54000)
	vendorID       = 0x10de
	deviceID       = 0x1b80
	deviceType     = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
	deviceName     = GeForce GTX 1080
```

simon13579, 2019-11-12:
Hello Martins，

Thanks, I think yes, it's because of VM. After I search the web according to your guess, I find out currently VM don't support vulkan. I will try on a real Ubuntu system, thanks.

Thanks.

closed 2019-11-12 by martins-mozeiko