lgsvl/simulator #479: Autoware and LGSVL on different PC

dohkim119, 2019-11-12:
What I'm trying to do is that trying to simulate Autoware behavior on LGSVL with different platform. 

So for example, Autoware is running on my laptop and LGSVL is run on my PC. 

I tried steps that written in LGSVL website but it seems not working.

So please give me some guidelines for doing this

Thank you.

martins-mozeiko, 2019-11-12:
When you were following instructions in our Autoware documentation which step did not work?

dohkim119, 2019-11-12:
I have modified cluster info and simlaution part as well. Then I clicked the play button but it keep shows me invalid message.

martins-mozeiko, 2019-11-12:
You should not change anything in clusters, this feature is unrelated to how you run AD stack. You should leave cluster at default values.

If you get invalid error message, check the log file. It will show more detailed error message: https://www.lgsvlsimulator.com/docs/faq/#where-are-unity-log-files-located

dohkim119, 2019-11-12:
I'm kinda confused here. Do I have to use the localhost (which is a default cluster) even though my Autoware is run on top of different IP?

martins-mozeiko, 2019-11-12:
Yes. You should use localhost. Cluster functionality has nothing to do with AD stack.
It is meant for distributed computing which is not yet implemented, we are still working on this feature.

Connection from simulator to AD stack is done via rosbridge. If you run AD stack and rosbridge on different machine, then you'll simply enter IP address of bridge connection when adding vehicle to simulation.

dohkim119, 2019-11-12:
Thank you so much. I figure it out.

closed 2019-11-12 by dohkim119

jjully, 2021-06-16:
Can I ask you how to success using different platform? I installed LGSVL simulator at windows 10 and autoware at ubuntu.
When I press the button lgsvl simulator at autoware, nothing happens.
I made a test at windows PC with bridge info: ```(IP of ubuntu):(port I opened)```
and cluster info is saved of windows PC
Is there anything that I have to download or do something at ubuntu for autoware? Do two computers have to has same files like map or something?
Sorry for long questions, not easy for beginner.
Thanks a lot to read :)