lgsvl/simulator #480: bug in MapLaneSection

kissofwind, 2019-11-12:
Hi, we found a pertential bug in MapLaneSection class.
the following code in `SetLaneData` function
```
                if (isOneWay.Value)
                {
                    edited[edited.Count-1].leftBoundType = LaneBoundaryType.CURB;
                }
                else
                {
                    edited[edited.Count-1].leftBoundType = LaneBoundaryType.DOUBLE_YELLOW;
                }
```
will cause a illegal access when list `edited` contains no elements, And this will happen when a road has only one lane. We have made a temparay fix about this bug avoiding execution of this part when edited has no elements, but we hope lgsvl can have a more elegant modification about this part of code.
Thanks

zelenkovsky, 2019-11-12:
Hello @Vincent-Tsui-GWM, thanks for finding.

We will take a look, @luqiang21 what do you think?
If that's an easy fix, we will push it together with our weekly code push by the end of the week.

BR
Dmitry Z.

luqiang21, 2019-11-12:
@Vincent-Tsui-GWM @zelenkovsky In our documentation, it says a MapLaneSection should not have less than 2 lanes. The purpose of MapLaneSection is to set correct neighboring relations among parrallel lanes. We will fix this by not to proceed if there is only one lane under a MapLaneSection. Thanks for pointing out.

closed 2020-04-08 by EricBoiseLGSVL