lgsvl/simulator #481: Environment Source Code

KrzysztofDabek, 2019-11-12:
Hi,

I have been working on modifications to LGSVL simulator and I see that the source for SingleLaneRoad, Shalun and CubeTown are available on github. Would you by any chance upload AutonomouStuff and BoregasAve too?

martins-mozeiko, 2019-11-12:
AutonomouStuff and BoregasAve maps uses assets under license that does not permit distributing them in source form. We can release these maps only as compiled binary assets.

That said, we will be releasing SanFrancisco map from old simulator (we are currently in process of upgrading it to HDRP materials). And we will be working on plugin mechanism to extend map with custom objects and behaviors at runtime.

KrzysztofDabek, 2019-11-13:
Okay. Thank you for that information.

closed 2019-11-13 by KrzysztofDabek