lgsvl/simulator #483: Apollo experiences latency when run on Python API

hyunjaecho94, 2019-11-12:
I have issued a potential latency issue in the https://github.com/lgsvl/PythonAPI/issues/13#issue-519475751, but I haven't gotten a response yet, so I am reposing it here to hopefully get an answer soon.

> Apollo does not avoid a simple collision. The modules in Module Controller all work (localization, perception, planning, traffic light, transform, control, routing, and prediction stay on once enabled). I am thinking this may be an issue with latency, and Apollo does not perceive an approaching vehicle until it's too late and hence results in a collision. If I manually pause (put on break) briefly, Apollo is able to perceive the approaching vehicle and stop itself. Could this be an issue of latency? If so, where could the latency be from?

martins-mozeiko, 2019-11-12:
What is typical framerate you get when running simulator? If Simulator is not rendering data fast enough it will start to delay some messages. We have seen that if Apollo receives data with too big latency or not frequent enough, it will not drive correctly. In ideal conditions you should have framerate at least 40, or better 50 and up.

Check the lidar & camera message publishing frequency. These are used for perception. Lidar should be ~10 and camera ~13-15. If these numbers are much smaller, then this is problem of Simulator performance on your computer.

hyunjaecho94, 2019-11-13:
How can I check the frequencies for lidar and camera and set the framework?

martins-mozeiko, 2019-11-13:
For CyberRT use cyber_monitor, for ROS based Apollo use `rostopic hz` command.

closed 2020-04-08 by EricBoiseLGSVL