lgsvl/simulator #484: Cannot load/read assets in Unity (Import FBX Errors) - Only empty world

RamoramaInteractive, 2019-11-12:
According to https://www.lgsvlsimulator.com/docs/assets/#check-asset-consistency the main repository does not contain any models, so you have to download the assets and add them into the 
folders within the 'external' folder. After I did it and opened Unity, didn't see any modells and I get many errors, that Unity has problems to read the files.

[image]

I also clicked on "simulator > check..." and I received following errors.

```
Checking...
ERROR: File '/Assets/Scripts/Editor/OdrSpiral.cs' does not have correct copyright header
ERROR: File '/Assets/Scripts/Editor/OpenDRIVE_1.4H.cs' does not have correct copyright header
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/Shaders/EnvironmentSimulation.shadergraph'
Done!

```

I have no idea, what is wrong with the Shaders or with Unity.
How can I fix these problems to see the 3D world environment?

martins-mozeiko, 2019-11-12:
Please check if you have [Git LFS](https://git-lfs.github.com/) installed. If any of those files in Console errors have tiny sizes (few hundred bytes), then git repository was cloned without Git LFS being available. Install git lfs first, before cloning, or fix repository afterwards by fetching & checking out lfs files: `git lfs fetch && git lfs checkout`

We use Git LFS for large file storage to save space when cloning git repository - it allows to download files only what are necessary in checkout, not they whole history.

RamoramaInteractive, 2019-11-14:
@martins-mozeiko I did it, but I receive new errors:

[image]

martins-mozeiko, 2019-11-14:
These are not real errors. Press Clear button and they will go away.
Unity likes to complain with errors about some of its internal workings (maybe due to fact that Linux platform is not very well supported). But these errors are harmless. Everything else still works.

RamoramaInteractive, 2019-11-22:
@martins-mozeiko 

I've added the environment and the vehicle into the external asset folder and build this project.
After starting the simulator I can't neither select a map nor the vehicle.

[image]
[image]

martins-mozeiko, 2019-11-22:
The error during asset bundle build happens because you do not have Linux support installed for your Unity. When we build asset bundles we are creating them platform independent - that means it includes parts that are needed for running on Linux, and parts running on Windows. You need to run Unity installer again and enable Linux support. After that the build should succeed.

RamoramaInteractive, 2019-11-23:
@martins-mozeiko: I reinstalled Unity 2019 1.10f1 and checked "Linux Support". I still receive the same error. :'( 

If I check the simulator I receive the same errors.
```
ERROR: File 'DampnessGeneratorSubGraph.shadersubgraph' with '.shadersubgraph' extension is not allowed inside '/Assets/Shaders'
ERROR: File 'EmissionSubGraph.shadersubgraph' with '.shadersubgraph' extension is not allowed inside '/Assets/Shaders'
ERROR: File 'IntensitySubGraph.shadersubgraph' with '.shadersubgraph' extension is not allowed inside '/Assets/Shaders'
ERROR: File 'RainDripGenerator.shadersubgraph' with '.shadersubgraph' extension is not allowed inside '/Assets/Shaders'
ERROR: File 'RainDropGenerator.shadersubgraph' with '.shadersubgraph' extension is not allowed inside '/Assets/Shaders'
ERROR: File '/Assets/Scripts/Editor/OdrSpiral.cs' does not have correct copyright header
ERROR: File '/Assets/Scripts/Editor/OpenDRIVE_1.4H.cs' does not have correct copyright header
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Jaguar2015XE.prefab'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Jaguar2015XEModel.fbx'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEBody.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEGlass.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEHeadLightLowCookie.png'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEIndicatorLeft.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEIndicatorReverse.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEIndicatorRight.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEInterior.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XELightBrake.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XELightHead.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEMetalRough.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEMetalSmooth.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEMirror.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEPlasticRough.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XEPlasticSmooth.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XETire.mat'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XETireAlbedo.png'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XETireEmissive.png'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XETireMask.png'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/External/Vehicles/Jaguar2015XE/Models/Materials/Jaguar2015XETireNormal.png'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/Shaders/EnvironmentSimulation02.shadergraph'
Done!
```

Git LFS is installed and after
```
$ git lfs install
    > Git LFS initialized.
```

I cloned the repository like in the description: https://www.lgsvlsimulator.com/docs/build-instructions/

RamoramaInteractive, 2019-11-23:
I also compiled the unity project and I can't even select the external Assets I've loaded.

martins-mozeiko, 2019-11-23:
These are very different errors. It seems you have saved CubeTown map with vehicle inside of it. While this is OK to do while you are developing something inside Unity, you should not have any EGO vehicles on the map when you are building it. That's why Check is showing these errors. That said - it does not prevent build. If bundle build fails, that means there is some different error. It is hard to predict what error could happen. Please provide full output from console log when build fails.

RamoramaInteractive, 2019-11-24:
@martins-mozeiko Yes, I've downloaded the Asset Bundles, but the problem still goes on.

Sure, here is the full console output:

```

1. System.Windows.Forms.dll assembly is referenced by user code, but is not supported on StandaloneWindows64 platform. Various failures might follow.
UnityEditor.BuildPipeline:BuildPlayer(BuildPlayerOptions)
Simulator.Editor.Build:RunPlayerBuild(BuildTarget, String, Boolean) (at Assets/Scripts/Editor/Build.cs:647)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:202)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

2. Player build succeeded!
UnityEngine.Debug:Log(Object)
Simulator.Editor.Build:RunPlayerBuild(BuildTarget, String, Boolean) (at Assets/Scripts/Editor/Build.cs:661)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:202)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

3.  Unrecognized assets cannot be included in AssetBundles: "Assets/External/Environments/CubeTown/Models/CubeTown/Materials/PavementHeight.png".
UnityEditor.BuildPipeline:BuildAssetBundles(String, AssetBundleBuild[], BuildAssetBundleOptions, BuildTarget)
Simulator.Editor.Build:RunAssetBundleBuild(String, List`1, List`1) (at Assets/Scripts/Editor/Build.cs:363)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:210)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

4. Unrecognized assets cannot be included in AssetBundles: "Assets/External/Environments/CubeTown/Models/CubeTown/Materials/PavementAlbedo.png".
UnityEditor.BuildPipeline:BuildAssetBundles(String, AssetBundleBuild[], BuildAssetBundleOptions, BuildTarget)
Simulator.Editor.Build:RunAssetBundleBuild(String, List`1, List`1) (at Assets/Scripts/Editor/Build.cs:363)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:210)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

5. Unrecognized assets cannot be included in AssetBundles: "Assets/External/Environments/CubeTown/Models/CubeTown/Materials/SignalTrafficVerticalEmissive.png".
UnityEditor.BuildPipeline:BuildAssetBundles(String, AssetBundleBuild[], BuildAssetBundleOptions, BuildTarget)
Simulator.Editor.Build:RunAssetBundleBuild(String, List`1, List`1) (at Assets/Scripts/Editor/Build.cs:363)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:210)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

6. Unrecognized assets cannot be included in AssetBundles: "Assets/External/Environments/CubeTown/Models/CubeTown/Materials/PavementNormal.png".
UnityEditor.BuildPipeline:BuildAssetBundles(String, AssetBundleBuild[], BuildAssetBundleOptions, BuildTarget)
Simulator.Editor.Build:RunAssetBundleBuild(String, List`1, List`1) (at Assets/Scripts/Editor/Build.cs:363)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:210)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

7.  Unrecognized assets cannot be included in AssetBundles: "Assets/External/Environments/CubeTown/Models/CubeTown/Materials/BlackSquare.jpg".
UnityEditor.BuildPipeline:BuildAssetBundles(String, AssetBundleBuild[], BuildAssetBundleOptions, BuildTarget)
Simulator.Editor.Build:RunAssetBundleBuild(String, List`1, List`1) (at Assets/Scripts/Editor/Build.cs:363)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:210)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

8. Unrecognized assets cannot be included in AssetBundles: "Assets/External/Environments/CubeTown/Models/CubeTown/Materials/BuildingEmission.png".
UnityEditor.BuildPipeline:BuildAssetBundles(String, AssetBundleBuild[], BuildAssetBundleOptions, BuildTarget)
Simulator.Editor.Build:RunAssetBundleBuild(String, List`1, List`1) (at Assets/Scripts/Editor/Build.cs:363)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:210)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

9. Unrecognized assets cannot be included in AssetBundles: "Assets/External/Environments/CubeTown/Models/CubeTown/Materials/BuildingsNormal.png".
UnityEditor.BuildPipeline:BuildAssetBundles(String, AssetBundleBuild[], BuildAssetBundleOptions, BuildTarget)
Simulator.Editor.Build:RunAssetBundleBuild(String, List`1, List`1) (at Assets/Scripts/Editor/Build.cs:363)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:210)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

10. Unrecognized assets cannot be included in AssetBundles: "Assets/External/Environments/CubeTown/Models/CubeTown/Materials/SignalTrafficVerticalAlbedo.png".
UnityEditor.BuildPipeline:BuildAssetBundles(String, AssetBundleBuild[], BuildAssetBundleOptions, BuildTarget)
Simulator.Editor.Build:RunAssetBundleBuild(String, List`1, List`1) (at Assets/Scripts/Editor/Build.cs:363)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:210)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

11. Unrecognized assets cannot be included in AssetBundles: "Assets/External/Environments/CubeTown/Models/CubeTown/Materials/SignTrafficStopAlbedo.png".
UnityEditor.BuildPipeline:BuildAssetBundles(String, AssetBundleBuild[], BuildAssetBundleOptions, BuildTarget)
Simulator.Editor.Build:RunAssetBundleBuild(String, List`1, List`1) (at Assets/Scripts/Editor/Build.cs:363)
Simulator.Editor.Build:OnGUI() (at Assets/Scripts/Editor/Build.cs:210)
[... 239 lines cut ...]
ICSharpCode.SharpZipLib.Zip.ZipFile.CommitUpdate () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
[... cut ...]
UnityEngine.UIElements.UIElementsUtility.DoDispatch (UnityEngine.UIElements.BaseVisualElementPanel panel) (at C:/buildslave/unity/build/Modules/UIElements/UIElementsUtility.cs:255)
UnityEngine.UIElements.UIElementsUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at C:/buildslave/unity/build/Modules/UIElements/UIElementsUtility.cs:78)
UnityEngine.GUIUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at C:/buildslave/unity/build/Modules/IMGUI/GUIUtility.cs:179)

```

martins-mozeiko, 2019-11-24:
Something is wrong with all files that have `Could not create asset from` error message. My guess would be on wrong git lfs checkout. Either Git LFS was not installed correctly, or you installed it only after you did "git clone" command - which means files are not updated.

Verify that these files are correctly checked out - the size should be >300 bytes. Or you can simply try opening .png images to see if they are actually images.

RamoramaInteractive, 2019-11-24:
@martins-mozeiko I've installed Git LFS. In my drive D:/ I opened Git Bash and I typed in
`git lfs install` and I received the output 
`Git LFS initialized`. Then I cloned the Repo:
`git clone --single-branch https://github.com/lgsvl/simulator.git`

Although I had done exactly what the instruction says, I've still receive these errors.
Some .png files only showed white backgrounds.
> 
> If you downloaded bundles, why are you building cube town and/or vehicle?

I thought I need it because I can't install the map and vehicle via the Web User Interface.

RamoramaInteractive, 2019-11-24:
@martins-mozeiko

RamoramaInteractive, 2019-11-24:
@martins-mozeiko I've installed Git LFS. In my drive D:/ I opened Git Bash and I typed in
`git lfs install` and I received the output 
`Git LFS initialized`. Then I cloned the repo:
`git clone --single-branch https://github.com/lgsvl/simulator.git`

Although I had done exactly what the instruction says, I've still receive some errors.

> 
> If you downloaded bundles, why are you building cube town and/or vehicle?

I thought I need it because I can't install the map and vehicle via the Web User Interface. My error.

I can build the simulator now, but I still receive some errors.
Here is the new console's output:

```
Rebuilding Library because the asset database could not be found!

System.Exception: Cannot load. Incorrect path: Packages/com.unity.render-pipelines.high-definition/Editor/DefaultScene/DefaultSceneRoot.prefab Null returned.
  at UnityEngine.Experimental.Rendering.HDPipeline.ResourceReloader.Load (System.String path, System.Type type, System.Boolean builtin) [0x00036] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipelineResources\ResourceReloader.cs:129 
  at UnityEngine.Experimental.Rendering.HDPipeline.ResourceReloader.SetAndLoadIfNull (System.Object container, System.Reflection.FieldInfo info, System.String path, System.Boolean builtin) [0x0000c] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipelineResources\ResourceReloader.cs:139 
  at UnityEngine.Experimental.Rendering.HDPipeline.ResourceReloader.ReloadAllNullIn (System.Object container) [0x00076] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipelineResources\ResourceReloader.cs:38 
  at UnityEngine.Experimental.Rendering.HDPipeline.HDRenderPipeline.UpgradeResourcesIfNeeded () [0x0009c] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipeline\HDRenderPipeline.cs:400 
  at UnityEngine.Experimental.Rendering.HDPipeline.HDRenderPipeline..ctor (UnityEngine.Experimental.Rendering.HDPipeline.HDRenderPipelineAsset asset) [0x0033c] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipeline\HDRenderPipeline.cs:262 
  at UnityEngine.Experimental.Rendering.HDPipeline.HDRenderPipelineAsset.CreatePipeline () [0x00004] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipeline\HDRenderPipelineAsset.cs:35 
UnityEditor.AssetPreviewUpdater:CreatePreviewForAsset(Object, Object[], String)

System.Exception: Cannot load. Incorrect path: Packages/com.unity.render-pipelines.high-definition/Editor/DefaultScene/DefaultSceneRoot.prefab Null returned.
  at UnityEngine.Experimental.Rendering.HDPipeline.ResourceReloader.Load (System.String path, System.Type type, System.Boolean builtin) [0x00036] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipelineResources\ResourceReloader.cs:129 
  at UnityEngine.Experimental.Rendering.HDPipeline.ResourceReloader.SetAndLoadIfNull (System.Object container, System.Reflection.FieldInfo info, System.String path, System.Boolean builtin) [0x0000c] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipelineResources\ResourceReloader.cs:139 
  at UnityEngine.Experimental.Rendering.HDPipeline.ResourceReloader.ReloadAllNullIn (System.Object container) [0x00076] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipelineResources\ResourceReloader.cs:38 
  at UnityEngine.Experimental.Rendering.HDPipeline.HDRenderPipeline.UpgradeResourcesIfNeeded () [0x0009c] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipeline\HDRenderPipeline.cs:400 
  at UnityEngine.Experimental.Rendering.HDPipeline.HDRenderPipeline..ctor (UnityEngine.Experimental.Rendering.HDPipeline.HDRenderPipelineAsset asset) [0x0033c] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipeline\HDRenderPipeline.cs:262 
  at UnityEngine.Experimental.Rendering.HDPipeline.HDRenderPipelineAsset.CreatePipeline () [0x00004] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipeline\HDRenderPipelineAsset.cs:35 
UnityEditor.AssetPreviewUpdater:CreatePreviewForAsset(Object, Object[], String)

System.Exception: Cannot load. Incorrect path: Packages/com.unity.render-pipelines.high-definition/Editor/DefaultScene/DefaultSceneRoot.prefab Null returned.
  at UnityEngine.Experimental.Rendering.HDPipeline.ResourceReloader.Load (System.String path, System.Type type, System.Boolean builtin) [0x00036] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipelineResources\ResourceReloader.cs:129 
  at UnityEngine.Experimental.Rendering.HDPipeline.ResourceReloader.SetAndLoadIfNull (System.Object container, System.Reflection.FieldInfo info, System.String path, System.Boolean builtin) [0x0000c] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipelineResources\ResourceReloader.cs:139 
  at UnityEngine.Experimental.Rendering.HDPipeline.ResourceReloader.ReloadAllNullIn (System.Object container) [0x00076] in D:\sim\simulator\Packages\com.unity.render-pipelines.high-definition\Runtime\RenderPipelineResources\ResourceReloader.cs:38 
[... cut ...]
Shader warning in 'HDRP/Decal': implicit truncation of vector type at Packages/com.unity.render-pipelines.high-definition/Runtime/Material/Decal/DecalData.hlsl(28) (on d3d11)

Compiling Vertex program with _EMISSIVEMAP _ALBEDOCONTRIBUTION
Platform defines: UNITY_ENABLE_REFLECTION_BUFFERS UNITY_USE_DITHER_MASK_FOR_ALPHABLENDED_SHADOWS UNITY_PBS_USE_BRDF1 UNITY_SPECCUBE_BOX_PROJECTION UNITY_SPECCUBE_BLENDING UNITY_ENABLE_DETAIL_NORMALMAP SHADER_API_DESKTOP UNITY_LIGHT_PROBE_PROXY_VOLUME UNITY_LIGHTMAP_FULL_HDR

```

martins-mozeiko, 2019-11-24:
There are no real errors here. This kind of output is expected for successful build.

As for building bundles - you need to do that only when you are creating new or modifying existing map/vehicle.

If all you want to do is to run existing map & vehicle, there is nothing to build.

1) Launch Loader scene, that can open WebUI. Or just simply run binary executable downloaded from Releases tab on GitHub (no need to open Unity at all).

2) Get map & vehicle AssetBundle URL from https://content.lgsvlsimulator.com/ (just copy to clipboard, don't download)

3) Open in WebUI maps and vehicles section and create new map/vehicle by pasting URL you copied. 
https://www.lgsvlsimulator.com/docs/maps-tab/#how-to-add-a-map
Now simulator will download binary assets and you'll be able to create simulation.

If you want to modify map/vehicle, only then you need to go through `Simulator -> Build` menu item.

RamoramaInteractive, 2019-11-25:
@martins-mozeiko Thank you very much. Now I'm able to manage the simulator in Unity.

closed 2019-11-25 by RamoramaInteractive