lgsvl/simulator #485: starting simulation multiple times in a python script

hyunjaecho94, 2019-11-12:
How can I run the simulator multiple times within a program  by using the Python API? For example, I first run the below snippet of code to start and load the simulator.
```
sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)
if sim.current_scene == "BorregasAve":
  sim.reset()
else:
  sim.load("BorregasAve")
```
Then I do some stuff. If I want to run the above snippet again later in the program, `sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)` runs fine, but sim.method() does not seem to run, where method() refers to the simulator class methods, such as `sim.current_scene`, `sim.load()`, `spawns = sim.get_spawn()` etc. I tried `sim.close()`, but it results in `runtimeerror event loop is closed`.

martins-mozeiko, 2019-11-12:
We never tried creating multiple instances of Simulator in one program. We'll need to check how that works.
Workaround for this would be create sim only once at startup, and then keep it around when you need to use it next time.

david-gwa, 2019-11-13:
what 's the purpose ?  do you need to reload a different map/world inside python script ?

closed 2020-04-08 by EricBoiseLGSVL