lgsvl/simulator #486: ImportError: No module named 'lgsvl'

hdt2020, 2019-11-12:
Hi I am very new to LGSVL, and was running through the PythonAPI Tutorial (https://www.lgsvlsimulator.com/docs/python-api/). 

I git cloned the the PythonAPI repo into the folder that is created once you extract the lgsvl zip file, however, when I run . `/quickstart/05-ego-drive-in-circle.py` I get an ImportError saying no such module exists. I did run `pip3 install --user -e .`  Would anyone know what is causing this error and how I can fix it? -Thanks

hdt2020, 2019-11-12:
I was able to solve this issue by running `sudo python setup.py install`

closed 2019-11-12 by hdt2020

martins-mozeiko, 2019-11-12:
The error was happening most likely because you have installed pip3 in multiple places. python3 from your system was using libraries from one folder, but pip3 was installing to different place. You should cleanup your python pip's. Or alternatively always specify full path when running pip3 and python:

    /usr/bin/pip3 ... 
    /usr/bin/python3 ....