lgsvl/simulator #488: run lg(2019.09)  in container

david-gwa, 2019-11-13:
following the discussion at [issue #382](https://github.com/lgsvl/simulator/issues/382),

it's easy to build lg based on [nvidia vulkan docker](https://gitlab.com/nvidia/container-images/vulkan), and  simple run.sh :

```
#!/usr/bin/env bash

xhost +

sudo nvidia-docker run  -it \
    -p 8080:8080 \
    -e DISPLAY=unix$DISPLAY \
    -v /tmp/LG\ Silicon\ Valley\ Lab/LGSVL\ Simulator/:/tmp/LG\ Silicon\ Valley\ Lab/LGSVL\ Simulator/ \
    -v /tmp/.X11-unix:/tmp/.X11-unix \
    --cap-add=SYS_PTRACE \
    --security-opt  seccomp=unconfined \
    simulator201909_lg-run  /bin/bash

``` 

 after running ./simulator in docker,  but when access the webUI from host brower, it has issues:  

[image]

the `Player.log` show:

```
Error: execlp(http://localhost:8080/) No such file or directory

(Filename: ./PlatformDependent/LinuxStandalone/URLUtility.cpp Line: 26)
```

can you guys give some help ?

Thanks
David

sm-auto, 2019-11-13:
Can you check 'ifconfig' in your docker command?

I think you should enable '--net=host' option

martins-mozeiko, 2019-11-13:
I'm not completely sure, but you might need to edit config.yml and put `*` for `hostname`. Otherwise it listens on localhost interface, which is not forward to host. `*` will make it to listen on all interfaces, including internal docker interface.

Alternative is to use `--net=host` for docker which will not isolate network and everything will run same as on host.

Error message in logfile probably is shown when you pressed button "Open Browser". Because there is no browser installed inside docker image.

Simulator is not using cuda, so having that in docker image is not necessary. It should be possible to create image based on NVidia OpenGL docker image.

david-gwa, 2019-11-13:
@sm-auto  and @martins-mozeiko ,     `--net=host` works it out.  thank you guys ! 

       Also - simulator is not using cuda, so having that in docker image is not necessary. It should be possible to create image based on NVidia OpenGL docker image.

the only reason has `cuda` is to use the existing nvidia docker image, will cut it off later.

david-gwa, 2019-11-13:
in this way, if further create a swarm-network,  it will lead to a lgsvl Docker cluster then ?

daohu527, 2019-11-21:
@david-gwa How do you build the docker? I run the nvidia vulkan docker and mount the build files in docker. Run the simulator in docker, then I get errors below:  
The screen:  
[image]

The error log:  
```
root@moonx:/home/test/build# cat /root/.config/unity3d/Editor/Player.log
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=2
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan error VK_ERROR_INCOMPATIBLE_DRIVER (-9) file: ./Runtime/GfxDevice/vulkan/VKContext.cpp, line: 333
Vulkan error./Runtime/GfxDevice/vulkan/VKContext.cpp:333
Vulkan detection: 0
No supported renderers found, exiting
 
(Filename: ./PlatformDependent/LinuxStandalone/main.cpp Line: 639)
```

david-gwa, 2019-11-22:
in your player.log it said `Vulkan detection: 0`,  I think that maybe the problem.

* check [the base vulakn image](https://gitlab.com/nvidia/container-images/vulkan)

* add [vulkan lib image](https://gitlab.com/nvidia/container-images/samples/tree/master/vulkan/ubuntu16.04) , if  vkcube works well, you should be good to run lgsvl 

I actually build lgsvl image based on these two.

david-gwa, 2019-11-22:
I also try to run  lgsvl image in docker swarm cluster,  check [issue](https://gitlab.com/nvidia/container-images/vulkan/issues/2),   it doesn't work well.

or it's a bad idea to run vulkan lgsvl in docker swarm,  do you guys have any suggestions ?

martins-mozeiko, 2019-11-22:
I think you are missing  `-v /tmp/.X11-unix:/tmp/.X11-unix` argument. This is required to be able to talk X11. `--env DISPLAY` is not enough.

daohu527, 2019-11-22:
@david-gwa Thanks for you help!
I find some problem and solve it.
```
1. The nvidia driver must be update to >= 418.39 with CUDA10.1. Then you run the [example](https://gitlab.com/nvidia/container-images/vulkan). If there is no error then the driver will be ok.   
2. If you not use "nvidia-docker" but "docker", you should add "--runtime=nvidia" option.  
```
Then I can run the simulator, but when I select the resolution to start, **my entire computer screen will be black, and log out.** Then there is a box with a system error to report.

david-gwa, 2019-11-22:
@daohu527 ,  

can you build and run  [vulkan sample](https://gitlab.com/nvidia/container-images/samples/blob/master/vulkan/ubuntu16.04/vkcube/Dockerfile) ?   to append  vulkan libs to your container

then check the [glxgears image](https://gitlab.com/nvidia/container-images/samples/tree/master/opengl/ubuntu16.04/glxgears) which will add openGL libs to your container

looks lgsvl  need both vulkan and openGL libs to build the container, I am not sure  if other related.

also as mentioned previously,  there needs a `--net=host` parameter

daohu527, 2019-11-22:
@david-gwa 
I do build from "nvidia/vulkan:1.1.121-cuda-10.1-alpha" which contains "OpenGL (glvnd 1.1) + Vulkan 1.1.121".
```
FROM nvidia/vulkan:1.1.121-cuda-10.1-alpha
```
Then I need to install some packages that the simulator needs.

When I run the emulator in the game, it looks like it’s flashing back. The log is below:  
```
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: count=59
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[... 108 lines cut ...]
  Thread -> id: 7fd9d6ffd700 -> priority: 1 
  Thread -> id: 7fd9d67fc700 -> priority: 1 
  Thread -> id: 7fd9d5ffb700 -> priority: 1 
  Thread -> id: 7fd9d57fa700 -> priority: 1 
  Thread -> id: 7fd9d4ff9700 -> priority: 1 
Opening in existing browser session.
Getting current user
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Fallback handler could not load library /home/07TmpSpace/CItest/docker/build/simulator_Data/Mono/libKernel32
Fallback handler could not load library /home/07TmpSpace/CItest/docker/build/simulator_Data/Mono/libKernel32.so
Fallback handler could not load library /home/07TmpSpace/CItest/docker/build/simulator_Data/Mono/Kernel32
Fallback handler could not load library /home/07TmpSpace/CItest/docker/build/simulator_Data/Mono/libKernel32
Fallback handler could not load library /home/07TmpSpace/CItest/docker/build/simulator_Data/Mono/libKernel32.so
Fallback handler could not load library /home/07TmpSpace/CItest/docker/build/simulator_Data/Mono/libKernel32
Vulkan PSO: Inconsistent pipeline cache header size[0], newDataSize[80118]
GarbageCollector disposing of ComputeBuffer. Please use ComputeBuffer.Release() or .Dispose() to manually release the buffer.
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Culling group was not disposed. You have to call Dispose explicitly from the main thread. This will likely result in a crash.
 
(Filename: /home/builduser/buildslave/unity/build/Runtime/Export/Camera/CullingGroup.bindings.cpp 
```

martins-mozeiko, 2019-11-22:
What do you mean by "flashing back"?

martins-mozeiko, 2019-11-23:
Here is a Dockerfile that creates minimal image (no CUDA) with Vulkan capabilities, and downloads & unpacks our simulator:

```
# install dependencies

FROM ubuntu:18.04 AS base

RUN set -ex \
  && apt-get update \
  && DEBIAN_FRONTEND=noninteractive apt-get upgrade -y \
  && DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends -y \
    unzip \
    ca-certificates \
    libx11-6 \
    libxau6 \
    libxcb1 \
    libxdmcp6 \
    libxext6 \
    libvulkan1 \
    libgl1 \
  && apt-get clean

ENV NVIDIA_VISIBLE_DEVICES all
ENV NVIDIA_DRIVER_CAPABILITIES graphics,display

ADD "https://gitlab.com/nvidia/container-images/vulkan/raw/master/nvidia_icd.json" /etc/vulkan/icd.d/nvidia_icd.json

# unzip LGSVL Simulator

FROM base AS unzipper

ARG LGSVL_SIMULATOR_VERSION=2019.11-rc1
ADD "https://github.com/lgsvl/simulator/releases/download/${LGSVL_SIMULATOR_VERSION}/lgsvlsimulator-linux64-${LGSVL_SIMULATOR_VERSION}.zip" /tmp/lgsvlsimulator.zip

RUN set -ex \
  && cd /tmp \
  && unzip "lgsvlsimulator.zip" \
  && mv "lgsvlsimulator-linux64-${LGSVL_SIMULATOR_VERSION}" lgsvlsimulator

# final image

FROM base

COPY --from=unzipper /tmp/lgsvlsimulator /opt/lgsvlsimulator

CMD /opt/lgsvlsimulator/simulator
```

Put it inside `Dockerfile` file and run `docker build -t lgsvlsimulator .` in same folder that will build `lgsvlsimulator` image.

To run it use following command:
```
docker run -ti \
     --runtime=nvidia \
     --net=host \
     -e DISPLAY \
     -e XAUTHORITY=/tmp/.Xauthority \
     -v ${XAUTHORITY}:/tmp/.Xauthority \
     -v /tmp/.X11-unix:/tmp/.X11-unix \
     -v lgsvlsimulator:/root/.config/unity3d \
     lgsvlsimulator
```

Replace `--runtime=nvidia` with `--gpus all` if you are on Docker >=19.03.

This will run simulator and will store persistent data in `lgsvlsimulator` volume (database, downloaded maps/vehicles, logfile). WebUI can be accessed on http://localhost:8080

This docker image should work on Ubuntu 18.04 and on ArchLinux.

david-gwa, 2019-11-26:
hey, @martins-mozeiko ,  a little more about this question

does the new lgsvl still have  a static configure,   so no need go to the server page to select  resolution and etc.

martins-mozeiko, 2019-11-26:
It is still displayed. But I you don't install `libgtk2.0-0` then this dialog won't be shown.

closed 2020-04-08 by EricBoiseLGSVL