lgsvl/simulator #490: BorregasAve is not available

Francisobiagwu, 2019-11-13:
I am currenty following the instruction outlined in the getting started with the pythonAPI. When I tried to run the sample code 05-ego-drive-in-circle, I get the following errors:

PythonAPI-2019.10-rc1/quickstart$ python3 05-ego-drive-in-circle.py 
Traceback (most recent call last):
  File "05-ego-drive-in-circle.py", line 15, in <module>
    sim.load("BorregasAve")
  File "/PythonAPI-2019.10-rc1/lgsvl/utils.py", line 21, in new_f
    return f(*args, **kwargs)
  File "/PythonAPI-2019.10-rc1/lgsvl/simulator.py", line 36, in load
    self.remote.command("simulator/load_scene", {"scene": scene, "seed": seed})
  File "/PythonAPI-2019.10-rc1/lgsvl/remote.py", line 68, in command
    raise Exception(data["error"])
Exception: Environment 'BorregasAve' is not available

Please, what does this error mean?

martins-mozeiko, 2019-11-13:
You need to first add it to the simulator. In the webui go to Maps section: https://www.lgsvlsimulator.com/docs/maps-tab/#how-to-add-a-map
And add the BorregasAve there. You can get link to map file from content website: https://content.lgsvlsimulator.com/

Francisobiagwu, 2019-11-13:
[image]

it is already there. but the error persists

martins-mozeiko, 2019-11-13:
The name is different. Python script looks up maps by name. You have added "Borregas Avenue", but Python script is asking for "BorregasAve".

Francisobiagwu, 2019-11-13:
Thanks. That fixed it.

closed 2019-11-13 by Francisobiagwu