lgsvl/simulator #491: Dreawview error

Francisobiagwu, 2019-11-13:
I am currently following the instruction on using the apollo-5.0. I selected. But, the apollo kept showing the components as FATAL and ERROR. Also, the Routing editing is grayed out

[image]
[image]
[image]

[image]

Francisobiagwu, 2019-11-13:
All the sensors are turned on. But the error persists. 
[image]

martins-mozeiko, 2019-11-13:
It is OK to see FATAL/ERROR messages in Components panel. Those are status for hardware sensors which Apollo uses. But Simulator simulates sensors on ROS/Cyber level, so Apollo thinks hardware is not working. You can safely ignore those.

The problem why localization is not working is that you are using Autoware sensors. You probably took car which was preconfigured for Autoware sensors. Switching just the bridge type is not enough, because Autoware and Apollo topics are different. You can get full list of sensors Apollo needs here: https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/#complete-json-configuration

And you don't need to "turn on" sensors. If they are added to vehicle in Json configuration, they will be working always. Clicking "eye" icon in Visualize panel inside Simulator only visualizes them for you on the screen.

Francisobiagwu, 2019-11-13:
Thanks. That worked.

closed 2019-11-13 by Francisobiagwu