lgsvl/simulator #492: HD Map Import coordinates

moenasrallah, 2019-11-13:
Upon import of an Apollo HD Map in v2019.10, I can't see where exactly the new map objects are in the scene. The transform of each map object imported seems to have respective UTM coordinates as the position. For example (500630.6, 0 , 254.1758) for a MapLane won't show until I move it to coordinates visible in my view port, such as 0,0,0.

I see a "Map Local Positions" field under the Map Line script attached to the object, but going through every single child will take a very long time to adjust and display manually.

Is there an easier way to do this? Maybe I'm missing something? Since this is new, I tried to look for documentation and haven't yet found it.

Thanks!

martins-mozeiko, 2019-11-13:
This is a bug in our importer code.
We have fixed this internally, and will push new code to github later this week.

Currently you can apply following changes to ApolloMapImporter, that should fix this issue.

    diff --git "a/Assets/Scripts/Editor/ApolloMapImporter.cs" "b/Assets/Scripts/Editor/ApolloMapImporter.cs"
    index 1968673e..6b448a6f 100644
    --- "a/Assets/Scripts/Editor/ApolloMapImporter.cs"
    +++ "b/Assets/Scripts/Editor/ApolloMapImporter.cs"
    @@ -191,6 +191,7 @@ using apollo.hdmap;
                 foreach (var point in double3List)
                 {
                     var vector3 = MapOrigin.FromNorthingEasting(point.y, point.x, false, false);
    +                vector3.x -= 500000;
                     vector3.y = (float)point.z;
                     vector3List.Add(vector3);
                 }
    @@ -819,6 +820,7 @@ using apollo.hdmap;
                 foreach (var point in signal.boundary.point)
                 {
                     var pos = MapOrigin.FromNorthingEasting(point.y, point.x, false, false);
    +                pos.x -= 500000;
                     position += pos;
                 }
                 position /= 4;

moenasrallah, 2019-11-15:
Thanks Martins. That moved it locally to the viewport successfully. 

I noticed that visually displaying the HD map in its entirety is an issue. Is there a way to show all HD map items without needing to click on them one by one, after enabling "Display Lane" (for example) ?

If I click Display Lane, and then click on that object, then it visually shows, but as soon as I click on anything else out of the very long list of map items, it goes away.

I modified the map lane script to change the display lane bool to true, and I still have to manually click on the item to have it visually show

I'm trying to line up the HD Map with our mesh, and its a little difficult. I went ahead and selected all the items to see the map, and i can get a sense of it, but its slow to navigate.

martins-mozeiko, 2019-11-15:
You shouldn't use Display Lane checkbox for this. I think we added for some kind of debugging purpose, but normally to work with annotations you should not use it.

Instead you should use `Simulator -> Annotate HD Map` tool window. Then either click View All to see all annotations, or View Selected to see only selected one subtree.

moenasrallah, 2019-11-15:
I see. I used that View All button before but the HD Map was off screen and I thought it didn't work 😅

Works great now, thank you guys for the hard work. Closed.

closed 2020-04-08 by EricBoiseLGSVL