lgsvl/simulator #493: Getting Data and Controlling Ego vehicle

Francisobiagwu, 2019-11-14:
I am running the simulator with Apollo. If I want to control the vehicle as well as collect sensor data using python, can I do that using the Python scripts located in the pythonAPI folder. Or what direction should I follow, must I run the simulator in API mode only?

martins-mozeiko, 2019-11-14:
If you want to use Python script, then simulation needs to run in API mode. API mode of course allows to connect EGO vehicle to AD stack via bridge. You do this by calling `agent.connect_bridge` method. You can see examples in PythonAPI repository in `NHTSA-sample-tests\Vehicle-Following` folder.

However note that collecting image/lidar data from Python API will make simulation to pause. Which means AD stack must support interruptions. Otherwise it may behave wrongly because it does not receive sensor messages often enough. Apollo/Autoware AD stacks do not like when GPS/IMU messages are not send frequently enough, the control will not work well.

The best way to collect data would be from AD stack itself. Apollo and Autoware can record incoming messages in bag file. Which later can be parsed and data extracted.

Francisobiagwu, 2019-11-14:
Thanks for the response. Please, is there a way to collect the data from Apollo on the fly without having to record it? I need to be able to collect data and perform some operation with in real-time

martins-mozeiko, 2019-11-14:
Yes, you can do that by writing ROS or Cyber node that receives messages and performs needed calculations. This node can be written in C++ or Python.

Francisobiagwu, 2019-11-14:
Please, do you have an example code in python?

martins-mozeiko, 2019-11-15:
Here's an example that will receive main camera image and decode it with OpenCV and print out its resoluition:

```
from __future__ import print_function

from cyber_py import cyber
from modules.drivers.proto.sensor_image_pb2 import CompressedImage

import numpy as np
import cv2

def on_image(image):
  print("Received image:")
  print("  format =", image.format)
  print("  data =", len(image.data), "bytes")

  arr = np.fromstring(image.data, np.uint8)
  img = cv2.imdecode(arr, cv2.CV_LOAD_IMAGE_COLOR)
  print("  size = {}x{}".format(img.shape[1], img.shape[0]))

if __name__ == '__main__':
  cyber.init()
  node = cyber.Node("MyNodeName")
  node.create_reader("/apollo/sensor/camera/front_6mm/image/compressed", CompressedImage, on_image)
  node.spin()
  cyber.shutdown()
```

You need to run this script inside Apollo container. Lidar and other sensors can be retrieved in similar way.

closed 2020-04-08 by EricBoiseLGSVL

VentiOvaltine, 2020-08-02:
I cannot 

> Here's an example that will receive main camera image and decode it with OpenCV and print out its resolution:
> 
> ```
> from __future__ import print_function
> 
> from cyber_py import cyber
> from modules.drivers.proto.sensor_image_pb2 import CompressedImage
> 
> import numpy as np
> import cv2
> 
> def on_image(image):
>   print("Received image:")
>   print("  format =", image.format)
>   print("  data =", len(image.data), "bytes")
> 
>   arr = np.fromstring(image.data, np.uint8)
>   img = cv2.imdecode(arr, cv2.CV_LOAD_IMAGE_COLOR)
>   print("  size = {}x{}".format(img.shape[1], img.shape[0]))
> 
> 
> if __name__ == '__main__':
>   cyber.init()
>   node = cyber.Node("MyNodeName")
>   node.create_reader("/apollo/sensor/camera/front_6mm/image/compressed", CompressedImage, on_image)
>   node.spin()
>   cyber.shutdown()
> ```
> 
> You need to run this script inside Apollo container. Lidar and other sensors can be retrieved in similar way.

Hi, I have the same question about reading sensor data, especially GPS sensor data. I tried your code but it cannot import cyber because `cyber.py` imports a package that does not exist named `_cyber_wrapper`. 
The code that imports `_cyber_wrapper` is in `apollo/cyber/python/cyber_py/cyber.py`
` _CYBER = importlib.import_module('_cyber_wrapper')`
Apollo version is 5.5.
Is there any suggestion?

martins-mozeiko, 2020-08-03:
Have you compiled Apollo 5.5? And are you running it in same docker environment?
If something does not with Apollo or CyberRT I suggest asking questions in Apollo community:  https://github.com/ApolloAuto/apollo/issues/

EricBoiseLGSVL, 2020-08-03:
Thanks @martins-mozeiko .
@VentiOvaltine Are you using Apollo 5.5 fork pre6? Issue #861 is about the issues with 5.5 master.