lgsvl/simulator #494: Vehicle not navigating to destination

Francisobiagwu, 2019-11-14:
These are the instruction under running he simulator with Apollo 5.0:
Navigate to the Route Editing tab.
Select a destination by clicking on a lane line and clicking Submit Route.
Watch the vehicle navigate to the destination.

However, when I select the destination as shown. After clicking on the 'send routinng request', the vehicle doesn't move. 

[image]
[image]
[image]

martins-mozeiko, 2019-11-14:
I think your EGO vehicle hit other vehicles, that's why it cannot move along the planned path.

Francisobiagwu, 2019-11-14:
I did a reset but the issue persists.

[image]

Francisobiagwu, 2019-11-14:
it works now.

closed 2019-11-14 by Francisobiagwu

khsrhhh, 2019-11-18:
> it works now. However the vehicle is not always smart. Sometimes, it keeps crashing into the woods.

Hi.
Now I'm facing same issue on Apollo 5.0.

Instruction sequence:
1. Set referenced ego car (Lincoln2017MKZ) and referenced map (Borregas Ave) in Dreamview
2. Turn all modules on except control module in Dreamview
3. Select end point (check the red line appears) and click 'Send Routing Request' button
4. Turn control module on in Dreamview
5. Noting happen (the green area does not appear)

Could you tell me how did you control car?

Thanks.

khsrhhh, 2019-11-19:
> Hi.
> Now I'm facing same issue on Apollo 5.0.
>
> Instruction sequence:
>
> 1. Set referenced ego car (Lincoln2017MKZ) and referenced map (Borregas Ave) in Dreamview
> 2. Turn all modules on except control [...]

Just enter up key in keyboard clears the issue.