lgsvl/simulator #495: Controlling different Ego vehicles with python

Francisobiagwu, 2019-11-14:
The WebUI allows the addition of multiple Ego vehicles to the scene. Please, is there a way for me to use for example 2 different python scripts to control the behavior of the vehicles simultaneously? As shown below:

 
[image]

martins-mozeiko, 2019-11-14:
You don't really need two Python scripts. You can create multiple EGO vehicles from same script. Just add more calls to `sim.add_agent` method. You can position them wherever you want and connect to different bridge instances - that will be controlled by different AD stacks.

daohu527, 2019-11-18:
Additionally, if I add two ego cars in the web UI. They cannot be reused and the names of the two cars must be different.

Francisobiagwu, 2019-11-22:
Thanks. Please, can you explain further how to connect to them using different bridge instances

martins-mozeiko, 2019-11-22:
Simply call `connect_bridge` on separate agent instances:

    agent1 = sim.add_agent("vehicle name 1", lgsvl.AgentType.EGO, state1)
    agent2 = sim.add_agent("vehicle name 2", lgsvl.AgentType.EGO, state2)

    agent1.connect_bridge("bridge address 1", 9090)
    agent2.connect_bridge("bridge address 2", 9090)

closed 2020-04-08 by EricBoiseLGSVL