lgsvl/simulator #496: execution of NPCs in the LG simulator is not deterministic

yl2280, 2019-11-14:
I found the execution of NPCs in the LG simulator is not
deterministic. Meaning if I repeat the same scenarios that I programmed
via PythonAPI (i.e., follow-lane, making lane change of NPCs), every
time I see a slightly different behaviors of NPCs (i.e., different
trajectories etc). Why? Can someone explain it for me and point me where
to look at in the code? (I assume there are stochastic process in the
simulator).

daviduhm, 2019-11-14:
Following lane and making lane change of NPCs via API should be deterministic, without an intervention of ego vehicle, because for each frame our simulator controls and moves each NPC with the same amount of distance in the same order based on user predefined seed. Could you share your PythonAPI code snippet so we can reproduce and look into this issue?

shalinmehtalgsvl, 2019-11-14:
@yl2280 

When you load a map with the API, are you setting the seed? If not, then determinism is not guaranteed.

https://www.lgsvlsimulator.com/docs/python-api/#simulation

yl2280, 2019-11-14:
@shalinmehtalgsvl Thank you so much! I am using the API with the lgsvl.2019.05, does it set the seed in the same way?

yl2280, 2019-11-14:
@shalinmehtalgsvl the behaviors of our NPCs are not random. We start them from the same location and specified their behavior in consecutive time period.

yl2280, 2019-11-14:
@daviduhm Thank you for your reply. We did use apollo3.5 in the simulator. Is that mater?

daviduhm, 2019-11-14:
Yes. AV stack like apollo is not deterministic unfortunately. If you are running apollo and having an ego vehicle controlled by apollo with NPCs around, then deterministic is not guaranteed.

yl2280, 2019-11-14:
@daviduhm so, can I have the conclusion that setting the seed when loading the map is not important in this case, because I did not let the NPCs run in random behavior.

martins-mozeiko, 2019-11-14:
The NPCs will behave randomly (at least it will look like that) because of interaction to EGO vehicle. Because Apollo AD stack is not deterministic - depending on how ROS/Cyber messages are sent and received, it will produce different control message. And simulator will move EGO vehicle only depending on when it receives this control message. So if in one run EGO vehicle decides turn a bit too much to right, or accelerate a bit faster than in different run, the NPC could decide different action, for example, because path ahead of it is not obstructed. Once one NPC changes it behavior, the others will follow, because they will see world differently - they will stop sooner or later to avoid collision and this effect will cascade further.

yl2280, 2019-11-14:
@martins-mozeiko Thank you so much. cuz I think there is a default seed set for simulator, I did not set the map seed. Will that also matter the behaviors of the NPCs? (I did setup control commands to NPCs, such as change lane, speed control, etc.)

martins-mozeiko, 2019-11-14:
Seed affects random generator for NPCs. This random generator is used to decide color of NPC, initial spawn position of NPC (in case you don't use Python API), and behavior to decide whether to turn left or right at intersection (if you use Python API `follow_closest_lane` call). If you set NPCs to follow waypoints, then the behavior should be 100% deterministic regardless of AD stack actions on EGO vehicle. Because when NPC follows waypoints, it follows them blindly without any collision detection with EGO vehicle.

Note that the deterministic behavior of NPCs was added in 2019.07 release of simulator. See changelog for more details: https://www.lgsvlsimulator.com/docs/changelog/

closed 2019-11-18 by yl2280