lgsvl/simulator #497: sensor object doesn't have s.data for camera and Lidar

Francisobiagwu, 2019-11-14:
I am trying to get access to the sensor data from the quickstart/06_save_camera_image.py. Instead of saving the image, I will like to access the image directly and then perform some operations with it, however, when I call the s.data, it seems like it is not implemented for both camera and lidar.

martins-mozeiko, 2019-11-14:
Because sending large binary data over websocket has large overhead, we decided that saving to disk will give better performance. You can simply load the file immediately after calling `save` method. You can increase saving performance by saving images/pcd files to memory disk, not on actual disk. On Linux you can do this by specifying `/dev/shm/file` location.

Francisobiagwu, 2019-11-14:
sensors = a.get_sensors()
    for s in sensors:
        if s.name == "Main Camera":
            # Camera and LIDAR sensors can save data to the specified file path
            s.save("main-camera.png", compression=0)
            s.save("main-camera.jpg", quality=75)
            break

    sim.run(time_limit=15)

These command saves only one image. I have tried to thread the sim.run() and the save image function, but that didn't work out. The same goes for multiprocessing. Please, how can I save image for the entire sim.run duration?

martins-mozeiko, 2019-11-14:
You would need to run simulation for short time, before saving again. And repeat that in a loop:

```
# find the camera sensor
for s in sensors:
  if s.name == "Main Camera":
    camera = s

num = 0

# repeat infinitely
while True:

  # save image
  s.save("main-camera{}.jpg".format(num), quality=75)
  num += 1

  # run simulation for a second
  sim.run(time_limit=1)
```

Francisobiagwu, 2019-11-14:
Thanks. I tried this and while it worked, it keeps slowing down the simulator. The simulator appears to move, and then stops, and then moves again.

closed 2020-10-06 by EricBoiseLGSVL