lgsvl/simulator #498: Rosbridge performance

Wang4758, 2019-11-15:
Dear Team,

We ran simulator version **2019.09** with **Apollo 3.0**. It happened a lot that the ego vehicle jitters in Dreamview. We did not see this behavior before with standalone Apollo. Do you know if it was because the performance of the rosbridge, or the localization module of Apollo?

Thanks in advance.

martins-mozeiko, 2019-11-15:
There could be multiple reasons. It could be rosbridge performance, it could be simulator performance, or Apollo performance.

To determine if this is rosbridge issue, check the CPU usage of rosbridge process. If it maxes out to 100% then that's definitely the bottleneck. Usually this happens when you send too much data.

To determine if this is Simulator performance issue, check what is rendering framerate. If it is too low, the Simulator cannot send data fast enough to Apollo, so it will not be able to determine correct control. Usually Simulator requires something like 40 frames per second. Ideally 50 or more.

Wang4758, 2019-11-18:
Thanks for you reply. I also saw your response in [this issue](https://github.com/lgsvl/simulator/issues/483#issuecomment-553163167) that the frequence of Lidar should be ~10 and camera ~13-15. Which frequency is the rendering framerate of simulator that should be ideally 50 or more?

martins-mozeiko, 2019-11-18:
You can use Geforce Experience from Nvidia (usually installed with driver), to enable showing rendering framerate. Here's a guide how to do that: https://www.youtube.com/watch?v=4knUcbnbFjI

closed 2020-04-08 by EricBoiseLGSVL