lgsvl/simulator #499: error display about localhost:8888

clouddd9, 2019-11-16:
when I build the lgsvl simulator using Unity3D and open the simulator, the localhost:8888 showed as follows:
[image]
 what's wrong with it? how can I fix it?

daohu527, 2019-11-18:
There may be several reasons for this:
1. Did you run “npm install“ in the WebUI and compile the WebUI in Unity?
2. In browser debug mode, check the console for error messages?

martins-mozeiko, 2019-11-18:
Try clearing browser cache, or doing full referesh - `Ctrl + Shift + R`. It might be that your browser cached some css files from previous versions for WebUI.

closed 2019-11-29 by clouddd9