lgsvl/simulator #501: Autoware and LGSVL

ningfeiwang, 2019-11-18:
Hello,

I follow the instruction here https://www.lgsvlsimulator.com/docs/autoware-instructions/ to build the LGSVL and Autoware together. When I go to step Click "Map" to load the launch file pertaining to the HD maps,  only Borregas shows 'OK'. I wonder whether I can use other maps like Shalun.

When I try to load Shalun, it seems the HD map is loaded. But I cannot find the car and cannot initialize the position. Could you please help me to handle such problems? Thank you very much.
[image]

martins-mozeiko, 2019-11-18:
Just to clarify - you are setting initial position of vehicle in rviz on this map? This is required for NDT localization, you need to provide EGO vehicle position. Sometimes it does not work on first try, so you need to repeat this process few times until it localizes.

ningfeiwang, 2019-12-16:
Thank you and sorry for the late reply. I tried it for several times but it still does not work (does not appear any car information in the above map). Do I need to open some computing component in that case like NDT mapping? Thanks.

closed 2020-04-08 by EricBoiseLGSVL