lgsvl/simulator #502: png files throw error in  Unity build - "File could not be read"

tkbala, 2019-11-18:
Hi,

I am trying to build the SingleLaneRoad scene from the unity editor. However, upon trying to build using "Simulator"->"Build"-"Build", I get the following error

[image]

martins-mozeiko, 2019-11-18:
Can you check if these files exist in Assets folder? And that they are reasonable - for example, png file should be an image.

tkbala, 2019-11-18:
Yes, the files exist. Here is a screenshot of the Jaguar2015XE folder containing the png files.

martins-mozeiko, 2019-11-18:
Where is screenshot?
And those files are real files? Can you open them to see image?

tkbala, 2019-11-18:
Ah! Here's the screenshot
[image]

The filesizes for images are low and the images do not open. Looks like it was a git lfs issue. I had downloaded the .zip of the git repo. I tried to git clone with git lfs enabled, and it seems to download the png files properly

closed 2019-11-18 by tkbala

martins-mozeiko, 2019-11-18:
Yeah, unfortunately GitHub does not package lfs files inside zip downloads.

dganzella, 2024-06-10:
IF anyone in the future has this same problem, which solved for me, on MacOS was: