lgsvl/simulator #504: screen black

yumianhuli2, 2019-11-19:
Hello！The simulation test is running , But unity has been stuck in the loading screen？What is the reason for this？
[image]

[image]

martins-mozeiko, 2019-11-19:
Can you please check log file to see if it has any error message? Log file locations: https://www.lgsvlsimulator.com/docs/faq/#where-are-unity-log-files-located

yumianhuli2, 2019-11-20:
Desktop is 1920 x 1080 @ 120 Hz
[Vulkan init] extensions: count=14
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=14
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: count=33
[Vulkan init] extensions: name=VK_KHR_swapchain, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_push_descriptor, enabled=0
[Vulkan init] extensions: name=VK_KHR_sampler_mirror_clamp_to_edge, enabled=1
[Vulkan init] extensions: name=VK_KHR_shader_draw_parameters, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHX_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHX_multiview, enabled=0
[Vulkan init] extensions: name=VK_EXT_discard_rectangles, enabled=0
[Vulkan init] extensions: name=VK_EXT_shader_subgroup_ballot, enabled=0
[Vulkan init] extensions: name=VK_EXT_shader_subgroup_vote, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_control, enabled=0
[Vulkan init] extensions: name=VK_NV_dedicated_allocation, enabled=0
[Vulkan init] extensions: name=VK_NV_glsl_shader, enabled=0
[Vulkan init] extensions: name=VK_NV_clip_space_w_scaling, enabled=0
[Vulkan init] extensions: name=VK_NV_sample_mask_override_coverage, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_array2, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_swizzle, enabled=0
[Vulkan init] extensions: name=VK_NV_geometry_shader_passthrough, enabled=0
[Vulkan init] extensions: name=VK_NVX_device_generated_commands, enabled=0
[Vulkan init] extensions: name=VK_NVX_multiview_per_view_attributes, enabled=0
Vulkan API version 1.0.49
Vulkan vendor=[NVIDIA] id=[10de]
[... 169 lines cut ...]
Vulkan PSO: Pipeline cache has not changed skipping save handle[47ba0f0]
GarbageCollector disposing of ComputeBuffer. Please use ComputeBuffer.Release() or .Dispose() to manually release the buffer.
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Culling group was not disposed. You have to call Dispose explicitly from the main thread. This will likely result in a crash.
 
(Filename: /home/builduser/buildslave/unity/build/Runtime/Export/Camera/CullingGroup.bindings.cpp Line: 70)

Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox

It's all about play.log!And the only "error" message seems ERROR: Shader Shader is not supported on this GPU (none of subshaders/fallbacks are suitable)WARNING: Shader Unsupported: 'Hidden/Nature/Terrain/Utilities' - Setting to default shader.
Plz help me check this, Thank you very much indeed!

david-gwa, 2019-11-26:
do you have   /tmp/LG\ Silicon\ Valley\ Lab/LGSVL\ Simulator/vulkan_pso_cache.bin ?   I don't understand the reason, but this cache file may helps

mmozeiko, 2019-11-26:
@david-gwa  That file is not a problem here. It is internal to Unity for caching Vulkan pipeline state object.

@yumianhuli2 I think the log file you posted is incomplete. Because first thing that happens when you press red play button in WebUI is [this line](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Web/Modules/SimulationsModule.cs#L474) - thus log file should contain `Starting simulation with id` text. But I don't see it here.

Last few lines in log file suggests you clicked on Clusters section in WebUI. Try first close simulator, delete whole log file, then run simulator again and only click red play button to start simulation

martins-mozeiko, 2019-11-26:
@david-gwa  That file is not a problem here. It is internal to Unity for caching Vulkan pipeline state object.

@yumianhuli2 I think the log file you posted is incomplete. Because first thing that happens when you press red play button in WebUI is [this line](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Web/Modules/SimulationsModule.cs#L474) - thus log file should contain `Starting simulation with id` text. But I don't see it here.

Last few lines in log file suggests you clicked on Clusters section in WebUI. Try first close simulator, delete whole log file, then run simulator again and only click red play button to start simulation

EricBoiseLGSVL, 2020-04-08:
@yumianhuli2 We are currently working on a fix for this issue.  The problem is Linux and LiDAR sensor GPU readback

yumianhuli2, 2020-04-09:
> The problem is Linux and LiDAR sensor GPU readback

ok！Could you tell me when the problem will be fixed?

EricBoiseLGSVL, 2020-04-09:
@yumianhuli2 we are hoping this month

yumianhuli2, 2020-04-10:
> @yumianhuli2 we are hoping this month

Nice!

EricBoiseLGSVL, 2020-04-16:
@yumianhuli2 Are you running Clusters still on this simulation?

yumianhuli2, 2020-04-17:
Could I understand that your team have fixed this bug?

EricBoiseLGSVL, 2020-04-17:
@yumianhuli2 We will have a release later this month that will resolve this

yumianhuli2, 2020-04-18:
> @yumianhuli2 We will have a release later this month that will resolve this

OK! I will wait for this!Thank U!

yumianhuli2, 2020-05-14:
> @yumianhuli2 We will have a release later this month that will resolve this

Hello！ Have you resolved this bug？

EricBoiseLGSVL, 2020-05-14:
@yumianhuli2 we plan on a release this month with this fix

yumianhuli2, 2020-05-15:
> @yumianhuli2 we plan on a release this month with this fix

very Nice！

yumianhuli2, 2020-06-09:
> @yumianhuli2 we plan on a release this month with this fix

Hi！Has this bug been fixed？

EricBoiseLGSVL, 2020-06-09:
@yumianhuli2 We have released 2020.05 and it should fix GPU read back errors on Linux.  Please let us know if you still have issues.  Are you able to run a simulation?

yumianhuli2, 2020-06-10:
> @yumianhuli2 We have released 2020.05 and it should fix GPU read back errors on Linux. Please let us know if you still have issues. Are you able to run a simulation?
Thank you！I will try ！

yumianhuli2, 2020-06-17:
> @yumianhuli2 We have released 2020.05 and it should fix GPU read back errors on Linux. Please let us know if you still have issues. Are you able to run a simulation?

OK，It works！

closed 2020-06-17 by EricBoiseLGSVL