lgsvl/simulator #508: Build the simulator from the Apollo HDmap

yumianhuli2, 2019-11-20:
If I want to build the simulator from the Apollo HDmap,Is there any relevant documentation for this process?
For example: 
step 1: find the button that imports the hdmap

step 2: import hdmap

step 3: generate hdmap

step 4: Import related vehicles

step 5: add sensors

This process is different from the current setup process on the web page.But we were able to get down to the basics and modify the simulator. Thank you!

martins-mozeiko, 2019-11-21:
1) Create scene by choosing `File -> New Scene`

2) Go to `Simulator -> Import HD Map`

3) Choose file where is your map (typically  in `base_map.bin` file)

4) Press Import button

5) Go to `Simulator -> Annotate HD Map` and press `View All` button to see annotations.

6) You'll need to adjust trigger bounds for each intersection, if you want NPCs to be able to navigate on map. Choose each intersection (under `Intersections` element in hierarchy) - and adjust Trigger Bounds X & Z sizes, you'll see how bounds look like in Scene view.

7) To make EGO vehicle drive on map you'll need to add something as ground - in simplest case it can be plane. Add it to scene and adjust position. Make sure it has collider added to it - this is what Unity uses for physics collisions.

Once these steps are done, you can try running scene directly in editor (you can choose which vehicle to create on map in `Simulator -> Development Settings`).

Or you can build map to bundle. To do that save it to proper location (`Assets/External/Environments/MAPNAME`) and choose `Simulator -> Build`. After bundle is built you can add it to web interface as usual. See following link with information about where to place map and how to build bundle: https://www.lgsvlsimulator.com/docs/assets/

yumianhuli2, 2019-11-22:
> 1. Create scene by choosing `File -> New Scene`
>
>     2. Go to `Simulator -> Import HD Map`
>
>     3. Choose file where is your map (typically  in `base_map.bin` file)
>
>     4. Press Import button [...]

Hello!
I have pulled newest version ,but I have not seen 
"Choose file where is your map (typically in base_map.bin file)"

martins-mozeiko, 2019-11-22:
Make sure you have pulled latest source code.
The file input choosing button/field is there:

[image]

closed 2020-04-08 by EricBoiseLGSVL