lgsvl/simulator #509: Couldn't find the button to import the apollo5 map

yumianhuli2, 2019-11-20:
Couldn't find the button to import the apollo5 map，plz see the picture below.
[image]

zelenkovsky, 2019-11-21:
Does the button "Import" on this screenshot opens "Open File" dialog?

yumianhuli2, 2019-11-21:
No, no dialog for "Open File"

martins-mozeiko, 2019-11-21:
We had a bug in code that prevented you to choose which file to import. This is fixed in latest release. Please pull latest code, then the import should work.

yumianhuli2, 2019-11-22:
> We had a bug in code that prevented you to choose which file to import. This is fixed in latest release. Please pull latest code, then the import should work.

Hello!
I have pulled newest version ,but I have not seen 
"Choose file where is your map (typically in base_map.bin file)"

martins-mozeiko, 2019-11-22:
Make sure you have pulled latest source code.
The file input choosing button/field is there:

[image]

closed 2020-04-08 by EricBoiseLGSVL