lgsvl/simulator #510: Communication Tutorial

yumianhuli2, 2019-11-20:
For example, if I want to communicate with cyber messages, can I have a simple example or an internal program to illustrate how to deliver messages and publish/subscribe topics from scratch?

daohu527, 2019-11-21:
You can refer to any sensor like "CanBusSensor.cs". Below is the approximate process:
```
1. Add a sensor in "Assets/Scripts/Sensors"
2. Add Dataformat in "Assets/Scripts/Bridge/Data"
3. Add code in "Assets/Scripts/Bridge/Cyber"
4. Add sensor in webUI with json
```

martins-mozeiko, 2019-11-21:
See following issue with more information how to publish new information from sensor: https://github.com/lgsvl/simulator/issues/339

And here is Python example code that shows how to subscribe to data in CyberRT: https://github.com/lgsvl/simulator/issues/493#issuecomment-554142705

closed 2020-04-08 by EricBoiseLGSVL