lgsvl/simulator #511: LidarSensor parameters in UnityEditor

KrzysztofDabek, 2019-11-20:
I have noticed that while using **UnityEditor** version of the LGSVL simulator.
The Lidar parameters are correctly loaded from the _json_ file (checked in debugging).
But after inspecting the **Lidar** object and _LidarSensor_ script the parameters are loaded default (or rather from serialized values). So the config loaded from _json_ file is changed automatically.
This makes it impossible to edit selected parameters keeping the rest as loaded.

martins-mozeiko, 2019-11-21:
I'm not sure what is the issue here? Are you saying parameters from json are overwriting default Unity serialized values in prefab? If yes, then this works as intended. We want this behavior. Default prefab values does not matter much when you create and use vehicle in simulator. We want sensor parameters be specified in web interface. Currently it is a bit ugly JSON text, but we are working on improved web interface that will allow to edit list of sensors and their parameters in nicer interface.

KrzysztofDabek, 2019-11-21:
Actually, it works the other way. The json parameters firstly overwrite script's values but when the GameObject is inspected, the default values are set for the lidar again.

Instead of sliders changed by the json file, we get the sensor changed by the sliders. The information from json file is lost

martins-mozeiko, 2019-11-21:
Ah, I see. You are talking about values inside Unity Editor.
We have fixed this in latest build, it will be pushed to GitHub soon.

martins-mozeiko, 2019-11-21:
New code is pushed to github.

closed 2020-04-08 by EricBoiseLGSVL