lgsvl/simulator #512: Sensor Data Types

arslansarp, 2019-11-20:
Can you share units of sensors  such that;

- Gps,latitude,longitude: degree or radian?,
- Gps altitude : is it meters or something else ?
- Imu euler angles : degree or radian ? 
- Imu angular velocity : is it radian/s or something else ?
- Imu accelaration : is it m/s^2 or something else ?
- Imu linear velocity : is it m/s or something else? 

Is GPS North East Down ?? If not, what is it?
Also can you share units of radar,lidar?
Thank you for your help...

martins-mozeiko, 2019-11-20:
GPS latitude, longitude is in degrees. Altitude is in meters.

IMU euler angles are in degrees. Acceleration in m/s^2, linear velocity in m/s.

Maps like BorregasAve and AutonomouStuff are oriented same way how they are in real location. In Unity coordinates positive Z direction points North, and positive X direction points East.

All lidar and radar measurements are in SI units - meters, seconds, m/s.

arslansarp, 2019-11-21:
thank you so much

closed 2019-11-21 by arslansarp