lgsvl/simulator #513: fakeboundary not allowing lane merge-help

ninjacoder0, 2019-11-20:
Hi,
Is there any way of not having "Fakeboundryline" as it does not allow me to merge the middle lane.  Ideally I would like the left lane to merge with the middle lane.

What is the best approach to correct them lane merge. 
Thank you all.

[image]

daohu527, 2019-11-21:
You can cut off the middle lane and then merge the right lane.
```
|
| \
```

ninjacoder0, 2019-11-21:
I would like to have two lane, (middle and right) drive ahead and have the left lane merge with the middle. 

Is this possible with this simulator. Im struggling with it. 
Thank you

[image]

daohu527, 2019-11-21:
I understand your problem. In my opinion, the left lane and middle lane merge parts will overlap a bit. Below is an example for this:  
  
[image]

ninjacoder0, 2019-11-21:
Yes by doing this the left lane overlaps, but does the fake boundary lanes cause a problems when doing this?  Specially for Apollo, it doesn't let you complete the autonomous drive as the boundary stops it?

luqiang21, 2019-11-21:
@ninjacoder0 I don't think the boundary line will stop Apollo from lane changing or merging. You need to set the boundary type correctly. And you need to make sure as in @daohu527's example, the both end points of the left lane and middle lane 1 should be overlapping with the start point of middle lane 2. That will ensure these three lanes are connected. This is similar for two lanes merging into one in an intersection.

daohu527, 2019-11-22:
I agree with @luqiang21 "This is similar for two lanes merging into one in an intersection."  

You can also treat it as a junction.   
Of course, you can also position the left lane as a virtual lane similar to the apollo junction did. I think this may be more complicated to understand, so I suggest you try the above operation first. I'm not sure if the current simulator supports the virtual lane.

closed 2020-04-08 by EricBoiseLGSVL