lgsvl/simulator #515: LGSVL-VPP (Vehicle Physics Pro) Integration

mehmetdogru, 2019-11-21:
Hi everyone,

I am currently trying to modify LGSVL vehicles with [VPP ](https://vehiclephysics.com/) dynamics and inputs.

We have completed the vehicle model using VPP and it works in Unity enviroment but when we try to import the model into LGSVL we are having errors because it doesnt have the same structure with VPP.

Is there anyone who has used LGSVL & VPP together and help us to import the model?

Regards,
Mehmet

EdyJ, 2019-11-22:
These are the instructions for integrating Vehicle Physics Pro into exiting projects:

https://vehiclephysics.com/user-guide/setting-up-vpp/#using-vpp-in-existing-projects

In summary, you simply need to copy or export the folder **Assets\Vehicle Physics Pro** from the VPP project to the LGSVL project. You could then drop a VPP vehicle prefab anywhere in LGSVL scenes, and it will work.

mehmetdogru, 2019-11-22:
Thank you for your answer. Eventhough you can drop a VPP vehicle prefab in the scene, you can not build the vehicles as assetbundles so it means you cannot import any VPP vehicles on WEB UI. The problem is packing the vehicle as an assetbundle so LGSVL would recognize it as a vehicle.

EdyJ, 2019-11-22:
Can you describe the nature and details of the problem? I've confirmed with other VPP clients that they use AssetBundles without issues.

mehmetdogru, 2019-11-22:
The vehicles within LGSVL contain the `VehicleDynamics` script which many other scripts in LGSVL access to.

And to create a vehicle in LGSVL whose vehicle dynamics is governed by VPP, I believe one would need to write a custom `VehicleDynamics` class that makes `VP Vehicle Controller` communicate with other LGSVL classes.

I have gone through all of the references within the `VehicleDynamics` class and here is the table I have assembled: https://drive.google.com/file/d/1Cwl_Zcz5KF89jc7F5lYwZlezjG5UiuH5/view?usp=drivesdk
(
Bottom right cyan colored list is the outside classes that access to the `VehicleDynamics` class.
Black colored list contains methods within the `VehicleDynamics` class.
Red colored list contains variables within the `VehicleDynamics` class.
)

Now I will start replacing the functions within the `VehicleDynamics` class to create the interface with `VP Vehicle Controller` class.

If you have a simpler way of doing this, please share.

EricBoiseLGSVL, 2019-11-22:
@mehmetdogru 

Yes, you are correct.  LGSVL Simulator requires VehicleDynamics.cs.  We are currently working on new features for dynamics and are moving VehicleDynamics.cs to a base class that just interacts with sensors.  Then you would use whatever dynamics model class you want that inherits from VehicleDynamics.  We hope to release this as soon as we can.  Vehicle Physics Pro is a great solution and we hope to make it easy for users to use software like this.

@EdyJ 

Thanks for your help.  Maybe we can schedule a time to chat about making VPP easier to integrate into LGSVL Simulator?

EdyJ, 2019-11-23:
@mehmetdogru
You may use the recently added VPVehicleToolkit component, which implements the most commonly used vehicle features such as getting speed, starting engine, etc. Indeed, the list of properties and methods exposed in this component is very similar to your table. The entire source code of this component is available here:

https://vehiclephysics.com/components/vehicle-toolkit/

@EricBoiseLGSVL 
Sure! Please write me to edy@vehiclephysics.com directly and we'll set up a time to chat.

mehmetdogru, 2019-11-29:
I modify "VehicleDynamics.cs" script and wrote an interface script named "MyVpp.cs" with @xmfcx. 

I am sharing modified "VehicleDynamics.cs" and "MyVpp.cs" below:

[MyVpp.zip](https://github.com/lgsvl/simulator/files/3905180/MyVpp.zip)
[VehicleDynamics.zip](https://github.com/lgsvl/simulator/files/3905181/VehicleDynamics.zip)

_Thank you for the help_ @EdyJ and @EricBoiseLGSVL.

closed 2019-11-29 by mehmetdogru