lgsvl/simulator #517: Camera: How to take depth images?

RamoramaInteractive, 2019-11-21:
According to the LGSVL Simulator Manual (Update April 2019) on Page 20 - **Features - GPU Accelerated LiDAR** there is a option to show depth images. 

How can I take images as depth images?

```
a = sim.add_agent("Lincoln2017MKZ (Apollo 5.0)", lgsvl.AgentType.EGO, state)

# get_sensors returns a list of sensors on the EGO vehicle
sensors = a.get_sensors()
for s in sensors:
  print(s.name)
  if s.name == "GPU":
    # Camera and LIDAR sensors can save data to the specified file path
    s.save("depth-camera.png", compression=0)
    s.save("depth-camera.jpg", quality=75)
    break
```

In the documentation I only found this: https://www.lgsvlsimulator.com/docs/sensor-json-options/#depth-camera There is no sensor name like "Depth Camera".

martins-mozeiko, 2019-11-21:
The sensors on your vehicle depends on how you have configured vehicle. You need to add sensors you want to vehicle in Web interface. It is done in this page: https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-change-the-configuration-of-a-vehicle

RamoramaInteractive, 2019-11-21:
@martins-mozeiko Thank you! I thought I should focus on the camera.

RamoramaInteractive, 2019-11-21:
@martins-mozeiko I've got a new Problem:
After adding the Depth Cam Sensor in the vehicle in the Web interface I changed
`s.name = "Main Camera" `
into 
`s.name = "Depth Camera"`,
but my console output is:

```
Traceback (most recent call last):
  File "C:\Users\user\Desktop\PythonAPI\quickstart\Test\06-save-camera-image.py", line 30, in <module>
    s.save("main-camera.jpg", compression=75)
  File "c:\users\user\desktop\pythonapi\lgsvl\utils.py", line 21, in new_f
    return f(*args, **kwargs)
  File "c:\users\user\desktop\pythonapi\lgsvl\sensor.py", line 79, in save
    "compression": compression,
  File "c:\users\user\desktop\pythonapi\lgsvl\remote.py", line 68, in command
    raise Exception(data["error"])
Exception: Sensor 'a08d1181-f8c6-46cb-ae73-4559fc869bf3' is not a camera sensor
```

martins-mozeiko, 2019-11-21:
Oh, sorry. I checked the code, it seems we did not support saving depth camera to image. This will require adding a bit of code to make this work. If you want to take a look then changes should be done to `SensorCameraSave.cs` and `DepthCameraSensor.cs` files.

RamoramaInteractive, 2019-11-21:
> 
> 
> Oh, sorry. I checked the code, it seems we did not support saving depth camera to image. This will require adding a bit of code to make this work. If you want to take a look then changes should be done to `SensorCameraSave.cs` and `DepthCameraSensor.cs` files.

Where can I find the files` SensorCameraSave.cs` and `DepthCameraSensor.cs` and what should I change?

martins-mozeiko, 2019-11-21:
https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Api/Commands/SensorCameraSave.cs

https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/DepthCameraSensor.cs

You could check how code looks like for ColorCameraSensor and add similar functionality for depth camera.

RamoramaInteractive, 2019-11-26:
From ColorCameraSensor I got this code:

```
 if (sensor is ColorCameraSensor)
                {
                    var camera = sensor as ColorCameraSensor;
                    bool result = camera.Save(path, quality, compression);
                    api.SendResult(result);
                }
```

Should I create a new Execute() method in DepthCameraSensor.cs or just paste some parts in an existing method?

martins-mozeiko, 2019-11-26:
You mean `Save` method? Yes, you would need to create similar Save method which other cameras have.

RamoramaInteractive, 2019-11-28:
In my Unity Simulator project I added in `DepthCameraSensor.cs` following function:

```
        public bool Save(string path, int quality, int compression)
        {
            CheckTexture();
            Camera.Render();
            var readback = AsyncGPUReadback.Request(Camera.targetTexture, 0, TextureFormat.RGBA32);
            readback.WaitForCompletion();

            if (readback.hasError)
            {
                Debug.Log("Failed to read GPU texture");
                return false;
            }

            Debug.Assert(readback.done);
            var data = readback.GetData<byte>();

            var bytes = new byte[16 * 1024 * 1024];
            int length;

            var ext = System.IO.Path.GetExtension(path).ToLower();

            if (ext == ".png")
            {
                length = PngEncoder.Encode(data, Width, Height, 4, compression, bytes);
            }
            else if (ext == ".jpeg" || ext == ".jpg")
            {
                length = JpegEncoder.Encode(data, Width, Height, 4, quality, bytes);
            }
            else
            {
                return false;
            }

            if (length > 0)
            {
                try
                {
                    using (var file = System.IO.File.Create(path))
                    {
                        file.Write(bytes, 0, length);
                    }
                    return true;
                }
                catch
                {
                }
            }

            return false;
        }
```

I have the problem, that I don't know how to take some code parts from  `SensorCameraSave.cs`, because the class can only inheritate from one class and you can only use one namespace.

My WebUI fails, if I use a code line like  `var api = ApiManager.Instance;` because of the missing namespace `Simulator.Api.Commands`.

martins-mozeiko, 2019-11-29:
I'm not sure what you mean by WebUI builds failing. This is C# code that has nothing to do with WebUI. 

You can add as many `using` statements in C# code as you need. If you need to use some class that is not in current namespace, you simply put using its namespace at top of file (for example, `using Simulator.Api.Commands;` ) This has nothing to do with class inheritance.

RamoramaInteractive, 2019-11-29:
> 
> 
> I'm not sure what you mean by WebUI builds failing. This is C# code that has nothing to do with WebUI.

I mean, I receive the error message after I clicked on WebUI.

Thank you for the advice. I will report back soon.

RamoramaInteractive, 2019-11-29:
Done! My error, there was no need for any methods in `DepthCameraSensor.cs`, only the save() method.

I had to add in `SensorCameraSave.cs` a else-if-condition with

```
                else if (sensor is DepthCameraSensor)
                {
                    var camera = sensor as DepthCameraSensor;
                    bool result = camera.Save(path, quality, compression);
                    api.SendResult(result);
                }
```

closed 2019-11-29 by RamoramaInteractive

Goolo2, 2020-09-07:
> Done! My error, there was no need for any methods in `DepthCameraSensor.cs`, only the save() method.
>
> I had to add in `SensorCameraSave.cs` a else-if-condition with
>
> ```
>                 else if [...]

Excuse me，I couldn't find the path of the file SensorCameraSave.cs，could you tell me where is it？Thank you very much！