lgsvl/simulator #519: Manual control using keyboard

Mikhail2017, 2019-11-24:
Hi. I recently installed lgsvl, but I'm not able to move forward. It's absolutely unclear from documentation which combination of keys should I use. Ignition doesnt work.

RamoramaInteractive, 2019-11-24:
You can find the shortcuts right here: https://www.lgsvlsimulator.com/docs/keyboard-shortcuts/
How do you start the simulator? Do you run it via Python API or simply started a certain map with a car?

martins-mozeiko, 2019-11-24:
@Mikhail2017 By move forward you mean manually control car with keyboard? Make sure you have [Keyboard Control](https://www.lgsvlsimulator.com/docs/sensor-json-options/#keyboard-control) sensor added to [vehicle](https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-change-the-configuration-of-a-vehicle). Also if you started [simulation](https://www.lgsvlsimulator.com/docs/simulations-tab/) in `Interactive Mode` you will need to press Play button at bottom left corner of screen, because by default time is stopped.

Mikhail2017, 2019-11-25:
Sorry, I've forgotten about parking brake.

closed 2019-11-25 by Mikhail2017