lgsvl/simulator #520: lgsvlsimulator-windows64-2019.10 memory leak

chenzhiwo, 2019-11-25:
[image]
- Windows 10.0.18362.
- Ryzen 1700X
- 16GB RAM
- GTX760 4GB VRAM
- API only simulation
Restart simulation in WebUI not release memory at all.

martins-mozeiko, 2019-11-25:
Does the memory increase more if you start and stop again, start and stop again, ... (repeat multiple times)?
I'm guessing that Unity simply keeps memory allocated because it can be reused for later operations.

chenzhiwo, 2019-11-26:
Memory increase over time till it run out of 16GB of memory.
In the log file I found ton's of following  message:
```
ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed.
(Filename: C:\buildslave\unity\build\Runtime/Shaders/ShaderImpl/ShaderState.cpp Line: 443)
```
Maybe unity add a ShaderProgram every loop?
By the way, it work perfectly in my 9th gen Intel platoform.

martins-mozeiko, 2019-11-27:
I'm not sure what could be issue.

I have not seen before error about "ShaderProgram is unsupported" and jobified rendering. You could try open project in Unity Editor and disable "Graphics Jobs" under project settings. But that will probably affect performance negatively.

Have you tried updating nvidia driver to latest? Maybe that would help.

DXR passes in HDRP/Lit shader are not an issue. These are used only for DXR RayTracing features which are not enabled in our project. So it is ok they they are not supported (DXR anyways works only with d3d12, not d3d11 which we use).

chenzhiwo, 2019-11-28:
Thanks for your help. The latest windows nvidia driver dosen't help, but it works well on linux. Upgrade GPU solve the problem.

closed 2019-11-28 by chenzhiwo