lgsvl/simulator #521: [Autoware] Vehicle could not stop after finishing the planned path in rviz

qmitw188, 2019-11-26:
Hello,
When vehicles reach the target position in rviz, vehicles got velocity is 0 from twist_cmd of VehicleCmd.
Simulator caculates the ADAccelInput from the difference between this velocity and ActualLinVel.

However, simulator reset ADAccelInput after 0.5s. Because Autoware did not send another VehicleCmd and velocity of vehicle did not decrease to 0 yet, the vehicle would not stop after finishing the planned path.

Could you kindly tell me why simulator reset ADAccelInput after 0.5s from getting VehicleCmd?
Related code: Update() in VehicleControlSensor.cs

```
    if (Time.time - LastControlUpdate >= 0.5)    // > 500ms
    {
        ADAccelInput = ADSteerInput = AccelInput = SteerInput = 0f;
    }
```
Any suggestions to fix this issue are also very appreciated.

EricBoiseLGSVL, 2019-11-27:
@qmitw188 

VehicleControlSensor This to prevent the vehicle accelerating or steering when the AD connection is lost or stopped.  You could apply brakes over a set amount of time after this call

closed 2020-04-08 by EricBoiseLGSVL

tekrajchhetri, 2020-04-14:
@EricBoiseLGSVL I have similar problem and also have asked the question in ros discussion forum https://answers.ros.org/question/349373/what-is-better-way-to-stop-vehicle-in-autoware/. Could you help on how can I apply brake using launch file or python script in autoware?

EricBoiseLGSVL, 2020-04-16:
@tekrajchhetri instead of setting ADAccelInput to 0, you can decrease over time.  We are currently working on updating our autoware features and should have a better solution soon.  This fix requires changing LGSVL Simulator source code and make a new build