lgsvl/simulator #522: Building WebUI

RamoramaInteractive, 2019-11-26:
My colleague has a similar problem with the Building WebUI like in issue https://github.com/lgsvl/simulator/issues/404 , but he is using Windows 7 and not Linux. The issue was not so helpful.

If he starts to build the WebUI, he receives an output like this:

[image]

The path of Node.js is already set.
What could be the source of the npm problem?

martins-mozeiko, 2019-11-26:
Can you try running `npm run pack-p` in commandline from simulator folder? Does it work then?

RamoramaInteractive, 2019-11-26:
> 
> 
> Can you try running `npm run pack-p` in commandline from simulator folder? Does it work then?

Sorry, this neither works. :(
Here is the output:
[image]

martins-mozeiko, 2019-11-26:
These seem to be only warnings. If this is all you see, then everything is OK.
You can check output files (main.css and main.js) generated in `dist` folder - if they are there, then the build succeeded.

closed 2020-04-08 by EricBoiseLGSVL

Gxer, 2020-05-09:
i have the same issue. 

> > Can you try running `npm run pack-p` in commandline from simulator folder? Does it work then?
> 
> Sorry, this neither works. :(
> Here is the output:
> [image]

Hi
i have the same issue exactly the same as yours.
Do you solve the problem?

EricBoiseLGSVL, 2020-05-11:
@Gxer In the log you posted it says Compiled with warnings.  This should be fine.  Does the Simulator run after the install? Do you have the dist folder described above?

twuser666, 2020-09-24:
> i have the same issue.
> 
> > > Can you try running `npm run pack-p` in commandline from simulator folder? Does it work then?
> > 
> > 
> > Sorry, this neither works. :(
> > Here is the output:
> > [image]
> 
> Hi
> i have the same issue exactly the same as yours.
> Do you solve the problem?

Hi, I have the same problem.
Do you solve the problem?

EricBoiseLGSVL, 2020-09-25:
It says compiled with warnings.  This is fine

twuser666, 2020-09-25:
> It says compiled with warnings. This is fine

Thanks a lot!