lgsvl/simulator #523: How to generate .cs files based on modified .proto files

clouddd9, 2019-11-27:
Hello, Now I want to change the message in chassis.proto and generate a new chassis.cs. I use protogen to generate the cs files from apollo *.proto. However, it seems that the protogen cannot import the dependent protos and so errors occurred when parsing.
Can you tell me how do you generate the protobuf in CSharp format? Thank you so much !

martins-mozeiko, 2019-11-27:
See the answers from David in #207 

Here's the command: https://github.com/lgsvl/simulator/issues/207#issuecomment-503816945

Note that you need to patch protobuf-net, otherwise it will generate incorrect C# code. You can see the change we did here: https://github.com/lgsvl/simulator/issues/207#issuecomment-504613674

clouddd9, 2019-11-27:
which OS did you build the protogen in ? windows or Ubuntu?

martins-mozeiko, 2019-11-27:
Both work fine. We have run it on Windows and also on Ubuntu with mono.

closed 2019-11-29 by clouddd9