lgsvl/simulator #526: Invalid: Out of date Map AssetBundle when building default Maps with Unity

lisaran, 2019-11-28:
Hi,

I've tried to build SingleLaneRoad / CubeTown with Unity and use it with current simulator version 2019.11-rc1.
When I add the Maps to the simulator I get: Invalid: Out of date Map AssetBundle. Please check content website for updated bundle or rebuild the bundle.
In Unity I get several warnings and two errors while building:
```
Shader error in 'PointCloudSolid.compute': '' : compilation terminated at kernel RemoveHidden at line 87 (on vulkan)
Shader error in 'PointCloudSolid.compute': 'constructor' : not enough data provided for construction at kernel RemoveHidden at line 87 (on vulkan)
```
I enabled windows and linux support in unity 2019.1.10f1.
I use Ubuntu 18.04 and libvulkan1 version is 1.1.114.0-1~gpu18.04.1.

Thanks for your help.

RamoramaInteractive, 2019-11-28:
Did you initialized Git-LFS before you cloned the simulator?

lisaran, 2019-11-28:
Yes I did. Building the simulator works fine. I‘ve only problems with the assets.

martins-mozeiko, 2019-11-29:
These errors about compute shader are normal. You can safely ignore it.

Error message about "out of date bundle" can happen if you use incompatible simulator version at runtime from one you used to build bundle. Make sure that you use exactly same version if simulator. For example, if you use 2019.11-rc1 binary, then make sure you use source code from `2019.11-rc1` tag to build maps and vehicles. Earlier source code of simulator will produce incompatible maps/vehicle bundles.

HDSFaust, 2019-12-16:
did you import the asset-repos via git inside your simulator-repo?

Fred159, 2019-12-20:
I just delete these "invalid ...bundle.." blocks. However, I regret doing this. I should have save the configuration of maps and vehicles. 
So I recommendation is create "new" and copy the appropriate configuration strings. Then it will be work correctly.

martins-mozeiko, 2019-12-20:
You don't need to delete out of date maps or bundles. If this is about maps/vehicles we distribute, then all you need to do is Edit and change URL to latest one from https://content.lgsvlsimulator.com/ website. If it is about your custom built ones and you have different file location, simply edit and change URL to correct file location.

Fred159, 2019-12-21:
> You don't need to delete out of date maps or bundles. If this is about maps/vehicles we distribute, then all you need to do is Edit and change URL to latest one from [...]

I have deleted these files, how to reset these files?

enisotomata, 2019-12-21:
i sometimes also have this error after adding some assets to my working and successfully built unity project and build again. and i can not find what is causing this. when i check file sizes to compare with previously working built (2.5 gb), error giving built is always small(500 mb). if i clean and delete everything in scene it starts to build again.

martins-mozeiko, 2019-12-23:
@Fred159 if you want to get all the default configurations Simulator creates on startup and you are OK removing everything else you have now, then simplest solution is to delete configuration folder of our Simulator, and run **binary** release. It will recreate everything with default values. On Linux this folder is in `~/.config/untiy3d/LG Silicon Valley Lab` location.

But you don't need to do that as you can always add all default maps and vehicles from our [content](https://content.lgsvlsimulator.com/) website. Simply copy&paste AssetBundle address when adding maps/vehicles.

Fred159, 2019-12-24:
> @Fred159 if you want to get all the default configurations Simulator creates on startup and you are OK removing everything else you have now, then simplest solution is to delete configuration folder [...]

Thank you so much.~

closed 2020-04-08 by EricBoiseLGSVL

amadeuszsz, 2020-12-09:
I got same error when I'm running simulator - "Invalid: Out of date AssetBundle, rebuild or download latest AssetBundle.". I followed the tutorial step by step  https://www.lgsvlsimulator.com/docs/build-instructions/
I have Unity 2019.3.3f1 as mentioned in tutorial with Windows support. With already built simulator (lgsvlsimulator-linux64-2020.06) everything works fine but there is a problem with my Unity build. Any suggestions?

EricBoiseLGSVL, 2020-12-09:
2020.06 release is using 2015.3.15f1.  It's in the release notes.

amadeuszsz, 2020-12-14:
> 2015.3.15f1

Did you mean 2019.3.15f1? In release notes is written _**Unity update to 2019.3.15f1**_
Anyway, still same issue with master branch & unity 2019.3.15f1:
Invalid: **Out of date AssetBundle, rebuild or download latest AssetBundle.**
Any idea?

EricBoiseLGSVL, 2020-12-14:
sorry yes 2019.  You will need to create a new asset bundle for the new Unity version.

amadeuszsz, 2020-12-15:
> sorry yes 2019. You will need to create a new asset bundle for the new Unity version.

Still wont work. I build SanFrancisco map from your repo using

rajeshjoshi8792, 2021-06-30:
i m getting same error. 'Out of date Map AssetBundle. Please check content website for updated bundle or rebuild the bundle."
please guide me how to solve this error. i tried to add "CubeTown" map. but after building that map unity and simulator crashing.

EricBoiseLGSVL, 2021-06-30:
- First off I highly recommend using our latest simulator version.
- If you are using 06 release tag on github, you need to create the map with 06 release tag in cubetown repo.
- Be sure to delete app data folder in case you accidentally downloaded an older map
```
C:\Users\USER\AppData\LocalLow\LGElectronics\SVLSimulator
```
- if you are having issues with unity building, delete the Library folder in the simulator root folder.  This will have unity re cache all shaders that sometimes get corrupted.

If this doesn't work, please provide Unity Editor logs to see what the errors are.  You can also paste them from the Unity Editor Console panel.