lgsvl/simulator #527: Unity executable 'Simulator' not running

Avps1, 2019-11-28:
Hi, 
I was following this page : https://github.com/lgsvl/simulator to install the lgsvl simulator. 
I followed the instructions till 'Unzip the downloaded folder and run the executable.'
But the executable 'Simulator' wont run! It shows a page where you can choose display (LOw/HIgh/Ultra,etc), but then nothing happens! 
The zip downloaded is lgsvlsimulator-linux64-2019.11-rc1.zip
I am on Ubuntu 18.04 .
Please help!

RamoramaInteractive, 2019-11-28:
Hi @Avps1 try to follow these instruction and clone this simulator: https://www.lgsvlsimulator.com/docs/build-instructions/

For this simulator there is recommanded a computer with
4 GHz Quad core CPU
Nvidia GTX 1080 (8GB memory).

martins-mozeiko, 2019-11-29:
@RamoramaInteractive You need to build simulator only if you intent to modify its functionality. No need to build if all you need to is to run existing maps & vehicles. Executable you can download from GitHub releases is working just fine. It is not defective.

@Avps1 Please check the log file for error messages: https://www.lgsvlsimulator.com/docs/faq/#where-are-unity-log-files-located
Most common error is that Vulkan userspace libraries are not installed. Check that you have `libvulkan1` installed.

RamoramaInteractive, 2019-11-29:
> 
> 
> @RamoramaInteractive You need to build simulator only if you intent to modify its functionality. No need to build if all you need to is to run existing maps & vehicles. Executable you can download from GitHub releases is working just fine. It is not defective.

By "cloning" I mean the simulator which you opened in unity. I didn't wrote anything about the models. Yes, of course, the executable works fine.

mehmetdogru, 2019-11-29:
@Avps1 

I dont know your pc hardware specs but in addition to @martins-mozeiko's comment you can try running simulator by choosing **Very Low** graphics quality. For some hardware, high graphics qualities are not supported as far as i've experienced.

Avps1, 2020-01-09:
I have tried to download `libvulkan1` library with the command `$ sudo apt install libvulkan1`.
This is the result.
`[sudo] password for abhi: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libvulkan1 is already the newest version (1.1.126.0-2~gpu18.04.1).
The following packages were automatically installed and are no longer required:
  freeglut3 freeglut3-dev libdrm-dev libgl1-mesa-dev libgles1 libglu1-mesa-dev libglvnd-core-dev libglvnd-dev libice-dev libllvm7 libnvidia-cfg1-418 libnvidia-common-418 libnvidia-compute-418
  libnvidia-decode-418 libnvidia-encode-418 libnvidia-fbc1-418 libnvidia-gl-418 libnvidia-ifr1-418 libopengl0 libpthread-stubs0-dev libsm-dev libx11-dev libx11-doc libx11-xcb-dev libxau-dev
  libxcb-dri2-0-dev libxcb-dri3-dev libxcb-glx0-dev libxcb-present-dev libxcb-randr0-dev libxcb-render0-dev libxcb-shape0-dev libxcb-sync-dev libxcb-xfixes0-dev libxcb1-dev libxdamage-dev libxdmcp-dev
  libxext-dev libxfixes-dev libxi-dev libxmu-dev libxmu-headers libxshmfence-dev libxt-dev libxxf86vm-dev mesa-common-dev x11proto-core-dev x11proto-damage-dev x11proto-dev x11proto-fixes-dev
  x11proto-input-dev x11proto-xext-dev x11proto-xf86vidmode-dev xorg-sgml-doctools xserver-xorg-video-nvidia-418 xtrans-dev
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
`
However the simulator still does not respond. By the way, the specs of my computer is presented below.
NVIDIA-SMI 430.64       Driver Version: 430.64       CUDA Version: 10.1     GeForce RTX2080

martins-mozeiko, 2020-01-09:
What does "not respond" means? Which part of simulator does not respond? GUI? Web interface? Vehicle control? And if something crashes please check log files first.

Avps1, 2020-01-09:
@martins-mozeiko 
Kindly look at my edited request above.
Basically the 'lgsvl simulator' window appears for a few seconds, and then disappears.

martins-mozeiko, 2020-01-10:
Please check the logfiles after it closes. It may have more informative message what is going on and why it closes.

Simulator should works with RTX 2080 GPU. We use Simulator with 2080 in our office.

Fred159, 2020-01-22:
> @martins-mozeiko
> Kindly look at my edited request above.
> Basically the 'lgsvl simulator' window appears for a few seconds, and then disappears.

In my case, When I installed the nvidia driver, simulator became works correctly.

Avps1, 2020-01-23:
Thank you for your response. This solves the problem.

closed 2020-01-23 by Avps1