lgsvl/simulator #528: vehicle simulation frequency

yumianhuli2, 2019-11-29:
In general, the frequency requirement of vehicle simulation is greater than 1000hz. How to solve this problem?E.G.This directly affects the calculation of imu frequency.Thank you！

martins-mozeiko, 2019-11-29:
Our current simulator is optimized for real-time simulation where 1000Hz is not a realistic number to achieve for physics simulation. If you need such simulation frequency that would require using non-real time simulation.

Our Python API allows to run simulation slower than real time, I suggest you start looking in our [sim.run](https://www.lgsvlsimulator.com/docs/python-api/#non-realtime-simulation) method. And to go higher than 100Hz (which is current rate of physics simulation) will require building custom version of simulator - you can set physics timestep in project properties: https://docs.unity3d.com/Manual/class-TimeManager.html

yumianhuli2, 2019-11-30:
> Our current simulator is optimized for real-time simulation where 1000Hz is not a realistic number to achieve for physics simulation. If you need such simulation frequency that would require using [...]

I heard that the maximum frequency of the car of phyx3/4 will not exceed 100hz, even if the physics timestep is raised, it will not exceed this limit

yumianhuli2, 2019-12-08:
> Our current simulator is optimized for real-time simulation where 1000Hz is not a realistic number to achieve for physics simulation. If you need such simulation frequency that would require using [...]

Hello! This is a quote from one of the most popular plugin in the unity asset store
https://assetstore.unity.com/packages/tools/physics/edy-s-vehicle-physics-403
Supports any physics timestep up to 0.06，This seems to contradict what you said
[image]

martins-mozeiko, 2019-12-08:
You are misunderstanding what 0.06 means there. What author is saying, is that his vehicle physics simulation code supports large timesteps - even **up to** 0.06 seconds (meaning 1/0.06 = 16Hz). Game physics simulation typically breaks down if timestep is too large. That's why unity by default uses 50Hz (0.02 sec), and we use 100Hz (0.01 sec) for more precision. Having large timetseps could give more performance, but it could lead to less accuracy if implemented poorly. VPP author says that his code supports even 0.06 timestep (and obviously smaller ones). But you are asking 0.001 which is 60 times smaller.

yumianhuli2, 2019-12-08:
> You are misunderstanding what 0.06 means there. What author is saying, is that his vehicle physics simulation code supports large timesteps - even **up to** 0.06 seconds (meaning 1/0.06 = 16Hz). [...]

OK，got it！

closed 2019-12-09 by martins-mozeiko